ros2/ros2cli #580: Inconsistent output of “ros2 param list”

DrakeProvost, 2020-12-19:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 20.04.1 LTS (running as WSL2 on Windows 10)
- Installation type:
  -  from binaries 
    - (using [this Docker container](https://github.com/NIURoverTeam/Dockerfiles/tree/master/ros2_foxy))
- Version or commit hash:
  -  Foxy
- DDS implementation:
  -  not sure
- Client library (if applicable):
  -  N/A

#### Steps to reproduce issue

I'm currently running ROS2 Foxy in a Docker container on WSL2. I'm going through the [ROS2 tutorials](https://index.ros.org/doc/ros2/Tutorials/Parameters/Understanding-ROS2-Parameters/#ros2params) so I have nodes `/turtlesim` and `/teleop_turtle` running. For some reason, the output of ros2 param list is inconsistent, even after rebooting the Docker container. Running `ros2 daemon status` before testing `ros2 param list` (as recommended [here](https://answers.ros.org/question/366094/inconsistent-output-of-ros2-param-list/)) does not resolve the issue.

#### Expected behavior
Every time I run `ros2 param list` with the two nodes I have running, I should see:
```
/teleop_turtle:
  scale_angular
  scale_linear
  use_sim_time
/turtlesim:
  background_b
  background_g
  background_r
  use_sim_time
```
#### Actual behavior
Here is me running `ros2 node/topic/service list` and then `ros2 param list` a couple times to show the inconsistency in output (of the 2 nodes running, the params printed correspond to either two, one, or zero of them, but it should always be two):
```
mddmprovost@ros2_foxy<DOCKER:a2bb8d3f7a78>:~ $ ros2 param list
mddmprovost@ros2_foxy<DOCKER:a2bb8d3f7a78>:~ $ rqt_graph
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-mddmprovost'
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
mddmprovost@ros2_foxy<DOCKER:a2bb8d3f7a78>:~ $ ros2 param list
/teleop_turtle:
  scale_angular
  scale_linear
  use_sim_time
mddmprovost@ros2_foxy<DOCKER:a2bb8d3f7a78>:~ $ ros2 node list
/teleop_turtle
/turtlesim
mddmprovost@ros2_foxy<DOCKER:a2bb8d3f7a78>:~ $ ros2 service list
/clear
/kill
/reset
/spawn
/teleop_turtle/describe_parameters
/teleop_turtle/get_parameter_types
/teleop_turtle/get_parameters
/teleop_turtle/list_parameters
/teleop_turtle/set_parameters
/teleop_turtle/set_parameters_atomically
/turtle1/set_pen
/turtle1/teleport_absolute
/turtle1/teleport_relative
/turtle2/set_pen
/turtle2/teleport_absolute
/turtle2/teleport_relative
/turtlesim/describe_parameters
/turtlesim/get_parameter_types
/turtlesim/get_parameters
/turtlesim/list_parameters
/turtlesim/set_parameters
/turtlesim/set_parameters_atomically
mddmprovost@ros2_foxy<DOCKER:a2bb8d3f7a78>:~ $ ros2 topic list
/parameter_events
/rosout
/turtle1/cmd_vel
/turtle1/color_sensor
/turtle1/pose
/turtle2/cmd_vel
/turtle2/color_sensor
/turtle2/pose
mddmprovost@ros2_foxy<DOCKER:a2bb8d3f7a78>:~ $ ros2 param list
mddmprovost@ros2_foxy<DOCKER:a2bb8d3f7a78>:~ $ ros2 param list
/teleop_turtle:
  scale_angular
  scale_linear
  use_sim_time
/turtlesim:
  background_b
  background_g
  background_r
  use_sim_time
mddmprovost@ros2_foxy<DOCKER:a2bb8d3f7a78>:~ $ ros2 param list
mddmprovost@ros2_foxy<DOCKER:a2bb8d3f7a78>:~ $ exit  #exit in other terminals too, reboot the docker container, re-run turtlesim_node and turtle_teleop_key
exit
mddmprovost@DESKTOP-21RFLUN:~/Dockerfiles/ros2_foxy.master $ ./attach.sh
[sudo] password for mddmprovost:
mddmprovost@ros2_foxy<DOCKER:0a7db511bf2b>:~ $ ros2 param list
/teleop_turtle:
  scale_angular
  scale_linear
  use_sim_time
/turtlesim:
  background_b
  background_g
  background_r
  use_sim_time
mddmprovost@ros2_foxy<DOCKER:0a7db511bf2b>:~ $ ros2 param list
mddmprovost@ros2_foxy<DOCKER:0a7db511bf2b>:~ $ ros2 param list
/turtlesim:
  background_b
  background_g
  background_r
  use_sim_time
mddmprovost@ros2_foxy<DOCKER:0a7db511bf2b>:~ $ ros2 param list
/teleop_turtle:
  scale_angular
  scale_linear
  use_sim_time
mddmprovost@ros2_foxy<DOCKER:0a7db511bf2b>:~ $
```
#### Additional information
Docker version:
```
Client: Docker Engine - Community
 Version:           19.03.13
 API version:       1.40
 Go version:        go1.13.15
 Git commit:        4484c46d9d
 Built:             Wed Sep 16 17:02:52 2020
 OS/Arch:           linux/amd64
 Experimental:      false

Server: Docker Engine - Community
 Engine:
  Version:          19.03.13
  API version:      1.40 (minimum version 1.12)
  Go version:       go1.13.15
  Git commit:       4484c46d9d
  Built:            Wed Sep 16 17:01:20 2020
  OS/Arch:          linux/amd64
  Experimental:     false
 containerd:
  Version:          1.3.7
  GitCommit:        8fba4e9a7d01810a393d5d25a3621dc101981175
 runc:
  Version:          1.0.0-rc10
  GitCommit:        dc9208a3303feef5b3839f4323d9beb36df0a9dd
 docker-init:
  Version:          0.18.0
  GitCommit:        fec3683
```

fujitatomoya, 2021-01-08:
I cannot reproduce this problem with following environment.

- Intel(R) Core(TM) i7-3770 CPU @ 3.40GHz
- docker version 19.03.8, build afacb8b7f0
- docker container ubuntu:20.04 (host is also ubuntu 20.04) with binding display and x socket
- ros2 rolling source code build
- start turtle sample node (https://index.ros.org/doc/ros2/Tutorials/Parameters/Understanding-ROS2-Parameters/#ros2params)
- use following command to see if any parameters are missing. (at least 100 times confirmed)

```
root@a6cfc30c581b:~/docker_ws/ros2_colcon# while true; do ros2 param list; ros2 topic list; ros2 node list; ros2 service list; done
```

audrow, 2021-03-19:
Hey @DrakeProvost, thanks for submitting the issue.

I tried it in a Foxy Docker container after installing `ros-foxy-turtlesim`, and I wasn't able to reproduce the issue. I ran @fujitatomoya's command above (thanks, @fujitatomoya) for several minutes and things seemed as expected. (I also ran `while true; do ros2 param list; done` for a few minutes to confirm I wasn't missing things in the output of `ros2 param list` afterwards.)

Do you have any other suggestions on reproducing this issue?

DrakeProvost, 2021-03-20:
Hi @audrow and @fujitatomoya, thanks for replying. I was able to reproduce this issue with the `osrf/ros:foxy-desktop` docker container. Please let me know if you need more environment information than what I've provided.

### Environment

* OS: Windows 10 Home, version 10.0.19041 Build 19041
* Processor: Intel(R) Core(TM) i5-5200U CPU @ 2.20GHz, 2201 Mhz, 2 Core(s), 4 Logical Processor(s)
* Docker version: 19.03.13, build 4484c46d9d (full version info in original post)
* Ubuntu version (in WSL2): Ubuntu 20.04.1 LTS
* ROS2: from docker container, see below

### My revised steps to reproduce

Open three terminals.

In terminal 1:
```bash
docker pull osrf/ros:foxy-desktop
xhost +local:root > /dev/null 2>&1 
docker run -it -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix:rw osrf/ros:foxy-desktop
# I'm now inside container
ros2 run turtlesim turtlesim_node # start turtlesim_node
```
In terminal 2:
```bash
docker exec -it <container> bash -c "source /opt/ros/foxy/setup.bash && bash"
# I'm now inside container
ros2 run turtlesim turtle_teleop_key # start turtle_teleop_key
```
In terminal 3:
```bash
docker exec -it <container> bash -c "source /opt/ros/foxy/setup.bash && bash"
# I'm now inside container
while true; do ros2 param list; echo "*** separator ***"; done
```
Sample output from terminal 3, showing that sometimes one or both of the nodes are excluded from `ros2 param list`:
```bash
/teleop_turtle:
  scale_angular
  scale_linear
  use_sim_time
/turtlesim:
  background_b
  background_g
  background_r
  use_sim_time
*** separator ***
/teleop_turtle:
  scale_angular
  scale_linear
  use_sim_time
/turtlesim:
  background_b
  background_g
  background_r
  use_sim_time
*** separator ***
*** separator ***
/teleop_turtle:
  scale_angular
  scale_linear
  use_sim_time
/turtlesim:
  background_b
  background_g
  background_r
  use_sim_time
*** separator ***
*** separator ***
/teleop_turtle:
  scale_angular
  scale_linear
  use_sim_time
*** separator ***
/teleop_turtle:
  scale_angular
  scale_linear
  use_sim_time
/turtlesim:
  background_b
  background_g
  background_r
  use_sim_time
*** separator ***
/turtlesim:
  background_b
  background_g
  background_r
  use_sim_time
*** separator ***
/teleop_turtle:
  scale_angular
  scale_linear
  use_sim_time
/turtlesim:
  background_b
  background_g
  background_r
  use_sim_time
*** separator ***
/teleop_turtle:
  scale_angular
  scale_linear
  use_sim_time
/turtlesim:
  background_b
  background_g
  background_r
  use_sim_time
*** separator ***
```

The output of `while true; do ros2 node list; ros2 topic list; ros2 service list; echo "*** separator ***"; done` is consistent, however (i.e. every iteration shows the below without any info being erroneously excluded):
```bash
/teleop_turtle
/turtlesim
/parameter_events
/rosout
/turtle1/cmd_vel
/turtle1/color_sensor
/turtle1/pose
/clear
/kill
/reset
/spawn
/teleop_turtle/describe_parameters
/teleop_turtle/get_parameter_types
/teleop_turtle/get_parameters
/teleop_turtle/list_parameters
/teleop_turtle/set_parameters
/teleop_turtle/set_parameters_atomically
/turtle1/set_pen
/turtle1/teleport_absolute
/turtle1/teleport_relative
/turtlesim/describe_parameters
/turtlesim/get_parameter_types
/turtlesim/get_parameters
/turtlesim/list_parameters
/turtlesim/set_parameters
/turtlesim/set_parameters_atomically
*** separator ***
```

fujitatomoya, 2021-03-26:
@DrakeProvost 

> Sample output from terminal 3, showing that sometimes one or both of the nodes are excluded from ros2 param list:

i was unable to have this problem on my Linux environment. sorry cannot find any windows machine with WSL2 though...

jacobperron, 2021-07-21:
I was able to reproduce I think the same issue with Rolling debs on Ubuntu 20.04 with `rmw_connextdds`. Cross-referencing https://github.com/ros2/ros2cli/issues/655, which may be related.

jacobperron, 2021-07-22:
I think this is fixed by #656. @DrakeProvost If you are able to try it out and confirm that would be great :slightly_smiling_face:

clalancette, 2022-02-03:
No response from the original reporter in over 6 months, so closing.  If you are still having a problem, feel free to reopen.

closed 2022-02-03 by clalancette