ros2/ros2cli #586: ros2 param list does not work with multiple rmw implementaion

fujitatomoya, 2021-01-21:
## Bug report

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - source code
- Version or commit hash:
  - https://github.com/ros2/ros2/commit/57b618fb80e6e93fc0323650378f1630a7d4773b
- DDS implementation:
  - multiple rmw implementation Fast-DDS and Cyclonedds
- Client library (if applicable):
  - rclcpp and rclpy

#### Steps to reproduce issue

- `talker` with cyclonedds and `ros2 param list` with fastdds

it seems that talker does not even receive the service request from client.

```
[Terminal - 1]
root@5c7a784cba7f:~/ros2_ws/colcon_ws# ros2 run demo_nodes_cpp talker
[INFO] [1611218728.615967478] [talker]: Publishing: 'Hello World: 1'
[INFO] [1611218729.615458541] [talker]: Publishing: 'Hello World: 2'
...<snip>

[Terminal - 2]
root@5c7a784cba7f:~/ros2_ws/colcon_ws# RMW_IMPLEMENTATION=rmw_fastrtps_cpp ros2 param list
Exception while calling service of node '/talker': None
```

- `talker` with fastdds and `ros2 param list` with cyclonedds

***this could be security issue since we can crash the node via parameter request.***

```
[Terminal - 1]
root@5c7a784cba7f:~/ros2_ws/colcon_ws# RMW_IMPLEMENTATION=rmw_fastrtps_cpp ros2 run demo_nodes_cpp talker
[INFO] [1611218853.091887892] [talker]: Publishing: 'Hello World: 1'
[INFO] [1611218854.091851082] [talker]: Publishing: 'Hello World: 2'
[INFO] [1611218855.091853219] [talker]: Publishing: 'Hello World: 3'
[INFO] [1611218856.091876166] [talker]: Publishing: 'Hello World: 4'
[INFO] [1611218857.091860947] [talker]: Publishing: 'Hello World: 5'
[INFO] [1611218858.091872255] [talker]: Publishing: 'Hello World: 6'
[INFO] [1611218859.091883693] [talker]: Publishing: 'Hello World: 7'
[INFO] [1611218860.091857624] [talker]: Publishing: 'Hello World: 8'
[INFO] [1611218861.091900999] [talker]: Publishing: 'Hello World: 9'
[INFO] [1611218862.091897370] [talker]: Publishing: 'Hello World: 10'
[INFO] [1611218863.091898792] [talker]: Publishing: 'Hello World: 11'
[INFO] [1611218864.091924685] [talker]: Publishing: 'Hello World: 12'
[INFO] [1611218865.091936687] [talker]: Publishing: 'Hello World: 13'
[INFO] [1611218866.091938711] [talker]: Publishing: 'Hello World: 14'
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

[Terminal - 2]
root@5c7a784cba7f:~/ros2_ws/colcon_ws# ros2 param list
root@5c7a784cba7f:~/ros2_ws/colcon_ws# ros2 param list
root@5c7a784cba7f:~/ros2_ws/colcon_ws# ros2 param list
Exception while calling service of node '/talker': None
```

#### Expected behavior

able to get parameter list for both cases

```
root@5c7a784cba7f:~/ros2_ws/colcon_ws# ros2 param list
/talker:
  use_sim_time
```

#### Actual behavior

either `unable to parameter list` or `application core crash`https://index.ros.org/doc/ros2/Tutorials/Working-with-multiple-RMW-implementations/#multiple-rmw-implementations

#### Additional information

according to https://index.ros.org/doc/ros2/Tutorials/Working-with-multiple-RMW-implementations/#multiple-rmw-implementations, there is no constraints / limitation between cyclonedds and fastdds.

clalancette, 2021-01-21:
Services do not work cross-vendor at the moment; since parameters are built on services, they won't work either.  This is a result of there being no standard way to do services in DDS.

That being said, the crash in the Fast-DDS talker when called from a Cyclone service is worrying.  That should not happen.  @EduPonz @MiguelCompany could you take a look?

MiguelCompany, 2021-01-21:
Seems like a deserialization exception we are not capturing. I will take a look.

fujitatomoya, 2021-01-22:
> Services do not work cross-vendor at the moment; since parameters are built on services, they won't work either. This is a result of there being no standard way to do services in DDS.

okay, i thought that should work. probably it would be nice to have description about the limitation.

MiguelCompany, 2021-01-22:
I have checked that the `std::bad_alloc` is thrown from Fast CDR code [here](https://github.com/eProsima/Fast-CDR/blob/f1c66e88ed25bc1f27b8465d7f9f6b22e62da7e3/include/fastcdr/Cdr.h#L2581).

There are two problems here:
1. Fast CDR should check the length before resizing the vector.
2. The rmw should capture Fast CDR exceptions and discard the received message

The first one is being solved by eProsima/Fast-CDR#98.

I will prepare a fix for the second one next week.

MiguelCompany, 2021-01-22:
@fujitatomoya @clalancette Please see ros2/rmw_fastrtps#505 for the fix on `rmw_fastrtps`

clalancette, 2021-01-22:
> okay, i thought that should work. probably it would be nice to have description about the limitation.

Agreed.  I thought we had something in the documentation, but I can't find it.  @fujitatomoya please feel free to add something to https://github.com/ros2/ros2_documentation explaining this limitation.

>     1. Fast CDR should check the length before resizing the vector.
> 
>     2. The rmw should capture Fast CDR exceptions and discard the received message

Great!  Thanks for the fast turnaround.

fujitatomoya, 2021-01-23:
@MiguelCompany 

wow, that was quick 💨  i will verify both of them to check if that works just in case.

@clalancette 

> please feel free to add something to https://github.com/ros2/ros2_documentation explaining this limitation.

i will consider this, and will come back to you with PR.

fujitatomoya, 2021-01-23:
@MiguelCompany with both fixes, i cannot reproduce the core dump. thanks for the fixes.

i will close this issue after,

- https://github.com/eProsima/Fast-CDR/pull/98 merged
- https://github.com/ros2/rmw_fastrtps/pull/505 merged
- documentation is updated

fujitatomoya, 2021-02-04:
since https://github.com/eProsima/Fast-CDR/pull/98 and https://github.com/ros2/rmw_fastrtps/pull/505 are merged, the rest is to update documentation which tells that services do not work cross-vendor rmw implementation. i will assign to this on me.

closed 2024-09-23 by clalancette