ros2/ros2cli #594: Try to autodetect matching QoS in `ros2 topic pub/echo`

nnmm, 2021-02-01:
## Feature request
It could be a great usability improvement if `ros2 topic echo` would make an effort to use a matching subscription QoS, and similarly for `pub`. For a previous discussion, see #156. The basic idea would be (talking about `echo` but the same applies to `pub`):

* When all publishers have the same/similar enough QoS, it could be possible to find a matching QoS profile and use that.
* When there are multiple QoS, the user has to make a choice.
* This should probably work in combination with the `--qos` flags, which would fix the specified parameters and let the others be deduced
* It should be made the default behavior, if there are no big downsides

But that only talks about a single point in time, whereas for a long-running `echo` command publishers can come and go. That raises the questions:
* Should the subscription "recalculate" its QoS when a new subscription/publisher on the other side joins? That would mean checking that there is a QoS that's consistent with the publishers and if that is not the current QoS, applying that to itself.
  * If the answer is "no", note that it would be very unintuitive to get different output when you run two publishers, depending on which one started first (`pub1`, `echo`, `pub2` vs `pub2`, `echo`, `pub1`). If I understand correctly, there are warnings about this now, which help somewhat.
  * If the answer is "yes", should it prompt the user when there is no consistent QoS, or just exit with a message?
* What should the QoS be when there is nothing to match with yet?
  * This is related to the previous question: If the QoS is recalculated, that "default" QoS profile is meaningless anyway

gonzodepedro, 2021-03-03:
I'm working on changes to implement partially some of this changes into ros2 topic echo. I have some questions:

1. Should this be circunscribed only to QoS profiles that come with ros2? or should it handle other possible combinations of QoS parameters?
2. With current default QoS parameters (best_effort, volatile), the echo subscriber should be able to receive messages from any publisher. But if the publisher is transient_local, then older messages won't be received. The idea would be to implement a logic that when all the publishers are transient_local, then make the subscriber transient_local as well, in order to receive older messages. Is that a step in the right direction?
3. I cannot find a reasonable way to detect when a new publisher joins, and to change QoS settings without risking to loose messages sent while the QoS change takes place. Any thoughts?

jacobperron, 2021-03-03:
> 1. Should this be circunscribed only to QoS profiles that come with ros2? or should it handle other possible combinations of QoS parameters?

I think it should handle any possible combination of QoS policies. I.e. we shouldn't limit ourselves to the preset profiles as users are free to create custom profiles.

> 2. With current default QoS parameters (best_effort, volatile), the echo subscriber should be able to receive messages from any publisher. But if the publisher is transient_local, then older [...]

I think that sounds reasonable.

> 3. I cannot find a reasonable way to detect when a new publisher joins, and to change QoS settings without risking to loose messages sent while the QoS change takes place. Any thoughts?

Maybe we don't have to detect when a new publisher joins exactly, rather we can detect when there's a new publisher with an incompatible QoS. There is a callback API that can let us know when a new incompatible publisher is detected ([see this  example](https://github.com/ros2/demos/blob/ed0299dc808f94b3c995dd4530680560e07226ff/quality_of_service_demo/rclpy/quality_of_service_demo_py/incompatible_qos.py)).

Another case to consider is when publishers drop from the graph. It's possible that if a publisher is gone, then the echo tool would be able to opt for a stricter QoS policy. E.g. consider the following:

- there are two publishers, one has best effort reliability and the other has reliable.
- "ros2 topic echo" subscribes with best effort since it matches with both publishers.
- the best effort publisher goes offline; now "ros2 topic echo" *could* change it's QoS to be reliable, which would be a better match, though not required.

Of course we still have a similar problem of detecting when there is a change to the ROS graph. I think we can get notified that *something* in the graph has changed, but are not told *what* changed. So, we'd have to query the state of the graph whenever something changes, and re-evaluated the QoS settings. I'm not sure how expensive this would be. On top of this, I *think* a graph listener API is only currently available in [rclcpp](https://github.com/ros2/rclcpp/blob/5a832e4db093a252f56aa854e0167d32465724f1/rclcpp/include/rclcpp/graph_listener.hpp) (there's no Python equivalent).

---

I would say that as a first step, we could query all publishers at startup and pick QoS based on that; be a strict as possible while maintaining compatibility. Then, any time we get an "incompatible QoS" event, re-calculate the QoS of the subscription. Listening to graph events to detect dropped publishers is probably best to implement separately as I think it is a bit more involved.

jacobperron, 2021-03-04:
Related rosbag2 ticket: https://github.com/ros2/rosbag2/issues/601

jacobperron, 2021-03-04:
Another potential issue we should consider is *how* we should decide if two QoS policies are compatible or not. The rules laid out [here](https://docs.ros.org/en/foxy/Concepts/About-Quality-of-Service-Settings.html#qos-compatibilities) are specific to DDS implementations. We probably want to make sure that the CLI tools work reasonably with different kinds of RMW implementations too. Maybe we could leverage the API being added in https://github.com/ros2/rclpy/pull/684 to help determine if a policy is valid (instead of hard-coding rules).

jacobperron, 2021-05-11:
Since https://github.com/ros2/ros2cli/pull/613, the `echo` verb makes an effort to match the QoS of existing publishers.

I think we could apply similar logic to the `pub` verb to resolve this ticket.

jacobperron, 2021-06-28:
Related PR https://github.com/ros2/ros2cli/pull/644

jacobperron, 2021-09-01:
The `pub` verb was resolved in https://github.com/ros2/ros2cli/pull/653