ros2/ros2cli #599: [ros2 action send_goal] autocompletion

doisyg, 2021-02-19:
## Feature request

#### Feature description
It would be incredibly useful to have auto-completion on the `ros2 action send_goal` cli tool (since there is no action_tool gui in ROS2, it is very tedious to debug actions at the moment)
The most needed auto-completion would be the goal message fields, but it would be nice to have it too for the action name and the action type.

clalancette, 2021-02-19:
This should work as of #570 .  I just tested it locally, and it worked for me here.  So I'll close this one out.

closed 2021-02-19 by clalancette

doisyg, 2021-02-19:
Amazing, it works, but with the same limitation for me as I documented here : https://github.com/ros2/ros2cli/issues/563
I need to put a quote + the first letter of the first field of the goal message for the auto-completion to work