ros2/ros2cli #601: Ros2 run doesn't indicate a segfault when it occurs.

amussell, 2021-02-23:
## Bug report
ros2 run doesn't print that a segfault occurred when one occurs making it look like the program just exited.
**Required Info:**

- Operating System:
  Ubuntu 20.04
- Installation type:
  apt
- Version or commit hash:
  foxy
- DDS implementation:
  default
- Client library (if applicable):
  rclcpp

#### Steps to reproduce issue
Run a program that segfualts with ros2 run.

#### Expected behavior
It should print a segmentation fault (core dump) when a segfault occurs.

audrow, 2021-03-12:
I was able to confirm this behavior. 

@amussell, feel free to open a PR for this and ping me as a reviewer.

fujitatomoya, 2021-09-07:
@audrow @amussell 

https://github.com/ros2/ros2cli/pull/639 can fix the problem and it's been merged. i will close this issue, feel free to open another ticket if we still have the problem.

closed 2021-09-07 by fujitatomoya