ros2/ros2cli #603: "ros2 node info --verbose" raises exception (rmw_cyclonedds_cpp)

jacobperron, 2021-02-24:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - 0.11.0
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Run a high frequency publisher, e.g. with image tools:
```
ros2 run image_tools cam2image --ros-args -p burger_mode:=true
```

In another shell, query verbose info about the topic

```
ros2 topic info --verbose /image
```

#### Expected behavior

We see output like the following:

```
Type: sensor_msgs/msg/Image

Publisher count: 1

Node name: cam2image
Node namespace: /
Topic type: sensor_msgs/msg/Image
Endpoint type: PUBLISHER
GID: c0.a8.00.73.00.01.aa.3f.00.00.00.01.80.01.08.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  Durability: VOLATILE
  Lifespan: 2147483649147483647 nanoseconds
  Deadline: 2147483649147483647 nanoseconds
  Liveliness: AUTOMATIC
  Liveliness lease duration: 2147483649147483647 nanoseconds

Subscription count: 0
```

#### Actual behavior

Exception is raised:

```
Type: sensor_msgs/msg/Image

Publisher count: 1

Traceback (most recent call last):
  File "/opt/ros/rolling/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.11.0', 'console_scripts', 'ros2')()
  File "/opt/ros/rolling/lib/python3.8/site-packages/ros2cli/cli.py", line 67, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/rolling/lib/python3.8/site-packages/ros2topic/command/topic.py", line 41, in main
    return extension.main(args=args)
  File "/opt/ros/rolling/lib/python3.8/site-packages/ros2topic/verb/info.py", line 62, in main
    for info in node.get_publishers_info_by_topic(topic_name):
  File "/usr/lib/python3.8/xmlrpc/client.py", line 1109, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python3.8/xmlrpc/client.py", line 1450, in __request
    response = self.__transport.request(
  File "/usr/lib/python3.8/xmlrpc/client.py", line 1153, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python3.8/xmlrpc/client.py", line 1169, in single_request
    return self.parse_response(resp)
  File "/usr/lib/python3.8/xmlrpc/client.py", line 1341, in parse_response
    return u.close()
  File "/usr/lib/python3.8/xmlrpc/client.py", line 655, in close
    raise Fault(**self._stack[0])
xmlrpc.client.Fault: <Fault 1: "<class 'OverflowError'>:int exceeds XML-RPC limits">
```

#### Additional information

With the CLI daemon stopped (`ros2 daemon stop`), we seem to get the exception after calling `ros2 topic info --verbose` two or more times. During the first invocation I notice a small delay before any output and it succeeds.

I can only reproduce the error with rmw_cyclonedds_cpp; I haven't been able to reproduce with rmw_fastrtps_cpp or rmw_connext_cpp.

jacobperron, 2021-02-24:
@eboasson Any ideas?

eboasson, 2021-02-24:
The cause is that the XML-RPC used by the daemon treats history depth as a signed integer and refuses to marshal a history depth ≥ 2\*\*31. The Cyclone RMW layer returns the history depth reported by Cyclone as an unsigned integer, and that happens to end up as 2\*\*32-1 for a keep-all history.

The depth is irrelevant for a keep-all history, so it can be dealt with by reporting 0 instead (https://github.com/eboasson/rmw_cyclonedds/commit/25715fdecd056438f98ac6f56090366f9d0158d0), but I think it is a bug that the marshalling can't represent all valid inputs. It is a `size_t` (https://github.com/ros2/rmw/blob/ae0553adb574c031343965195ec4a73039e62389/rmw/include/rmw/types.h#L441), so I guess it should accept any unsigned 64-bit integer.

clalancette, 2021-04-19:
I did some research here, and it turns out that XML-RPC only supports 4-byte signed integers: http://xmlrpc.com/spec.md .

The generally recommended workaround is to encode parameters as strings (and then do the conversion back and forth to an integer), or not use XML-RPC at all.  There is also the workaround that @eboasson suggested.  Given where we are in Galactic, moving away from XML-RPC isn't in the cards.  And while we could probably change the XML-RPC code to marshal back and forth from strings, I think it is simpler to just fix this in `rmw_cyclonedds`.  Tomorrow I'll go ahead and open a PR for that and add a comment on why that is necessary.

closed 2021-04-21 by clalancette