ros2/ros2cli #607: ros2 commandline message autocompletion needs reformatting to be usable

shonigmann, 2021-03-10:
## Bug report

**Required Info:**

- Operating System:
  - 
  - Ubuntu 20.04
- Installation type:
  - 
  - From binaries (for everything that I think is relevant to this issue, at least)
- Version or commit hash:
  - ros-foxy-ros2cli : 0.9.8-1focal.20210225.220420
  - ros-foxy-ros2cli-common-extensions : 0.1.0-1focal.20210225.235201

#### Steps to reproduce issue

Listener.py
```
import rclpy
from rclpy.node import Node

from std_msgs.msg import Float64MultiArray

class MinimalSubscriber(Node):

    def __init__(self):
        super().__init__('minimal_subscriber')
        self.subscription = self.create_subscription(
            Float64MultiArray,
            'topic',
            self.listener_callback,
            10)
        self.subscription  # prevent unused variable warning

    def listener_callback(self, msg):
        self.get_logger().info('I heard: "%s"' % msg.data)

def main(args=None):
    rclpy.init(args=args)

    minimal_subscriber = MinimalSubscriber()

    rclpy.spin(minimal_subscriber)

    # Destroy the node explicitly
    # (optional - otherwise it will be done automatically
    # when the garbage collector destroys the node object)
    minimal_subscriber.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

Terminal 1:
```
python3 Listener.py
```

Terminal 2:
```
ros2 topic pub /topic std_msgs/msg/Float64MultiArray layout*tab autocomplete*
```
#### Expected behavior
Terminal 2 autocompletes the command with a template, similar to that in ROS1, that can be easily populated (e.g. simply adding values to data[]) and run.

#### Actual behavior
```
ros2 topic pub /topic std_msgs/msg/Float64MultiArray layout:\
\ \ dim:\ []\
\ \ data_offset:\ 0\
data:\ [1.0, 2.0]\ 
usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...
ros2: error: unrecognized arguments: 2.0] 
```
#### Additional information
The only way I have successfully been able to publish a message of this type is by taking the fields suggested by the autocomplete and manually converting to a yaml dict string (.e.g.:
```
ros2 topic pub /topic std_msgs/msg/Float64MultiArray "{layout: {dim: [], data_offset: 0}, data: [1.0, -1.0]}"
```

I'm not sure if this is a bug or a feature request or if I'm just doing something wrong, so any feedback would be great

Thanks!

clalancette, 2021-03-10:
This was fixed on the `master` branch in #570 .  I think we just need to backport that to Foxy.  @shonigmann are you interested in taking a crack at backporting it and opening a PR against the `foxy` branch?

shonigmann, 2021-03-10:
Thanks for the quick response - that's good to know. 

I can try to take a crack at backporting, though I'm not sure I'll have time until the weekend

fujitatomoya, 2021-03-14:
@shonigmann Just FYI, i created backport PR https://github.com/ros2/ros2cli/pull/609

shonigmann, 2021-03-14:
@fujitatomoya amazing, thanks!

closed 2021-03-14 by shonigmann