ros2/ros2cli #608: :farmer: ros2param dump and load tests failing

tfoote, 2021-03-10:
## Bug report

rosparam dump and load tests are crashing on linux and osx in our nightly builds.

**Required Info:**

- Operating System:
  - Linux, and OSX
- Installation type:
  - nightly CI
- Version or commit hash:
  - master
- DDS implementation:
  - rmw_cyclonedds_cpp

#### Steps to reproduce issue

Tests with the failures last night:

* https://ci.ros2.org/job/nightly_linux_debug/1863/testReport/
* https://ci.ros2.org/job/nightly_linux_release/1849/testReport/
* https://ci.ros2.org/job/nightly_linux_repeated/2221/testReport/
* https://ci.ros2.org/job/nightly_linux-aarch64_debug/1536/testReport/
* https://ci.ros2.org/job/nightly_linux-aarch64_release/1502/testReport/
* https://ci.ros2.org/job/nightly_linux-aarch64_repeated/1528/testReport/
* https://ci.ros2.org/job/nightly_osx_debug/1955/testReport/
* https://ci.ros2.org/job/nightly_osx_release/1992/testReport/
* https://ci.ros2.org/job/nightly_osx_repeated/2277/testReport/
* 

#### Expected behavior

Test Passes

#### Actual behavior

```
FAIL: test_verb_load.TestVerbDump.test_verb_load[rmw_cyclonedds_cpp]
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jenkins-agent/workspace/nightly_linux-aarch64_debug/ws/src/ros2/ros2cli/ros2param/test/test_verb_load.py", line 290, in test_verb_load
    assert param_dump_command.exit_code == launch_testing.asserts.EXIT_OK
AssertionError: assert -11 == 0
 +  where -11 = <launch_testing.tools.process.ProcessProxy object at 0xffff7c449760>.exit_code
 +  and   0 = <module 'launch_testing.asserts' from '/home/jenkins-agent/workspace/nightly_linux-aarch64_debug/ws/install/launch_testing/lib/python3.8/site-packages/launch_testing/asserts/__init__.py'>.EXIT_OK
 +    where <module 'launch_testing.asserts' from '/home/jenkins-agent/workspace/nightly_linux-aarch64_debug/ws/install/launch_testing/lib/python3.8/site-packages/launch_testing/asserts/__init__.py'> = launch_testing.asserts
```

clalancette, 2021-03-10:
So I was able to reproduce this locally.  When I backed out all of the changes that went into rclpy yesterday and today, it went away.  That means that this is likely one of ros2/rclpy#703, ros2/rclpy#696, ros2/rclpy#701, or ros2/rclpy#702 .  @sloretz @cottsay Can you please take a look?

cottsay, 2021-03-11:
Regression is definitely caused by ros2/rclpy#701. I opened a PR to revert the change, which I'll merge before nightlies unless I find the bug: ros2/rclpy#710.

closed 2021-03-11 by cottsay