ros2/ros2cli #614: Topic: Pub and sub missing QOS options

matthews-jca, 2021-03-25:
----
## Feature request

#### Feature description
The topic verb, both pub and sub, don't allow for specification of the duration and liveliness QOS profile settings

https://github.com/ros2/ros2cli/blob/master/ros2topic/ros2topic/api/__init__.py#L161

audrow, 2021-05-07:
Hi @matthews-jca, thanks for creating the issue. I assume you mean the verb `echo` instead of `sub`, right?

This would be a nice thing to add. If you can open a PR for this, feel free to @ me as a reviewer or if you have any questions.

matthews-jca, 2021-05-07:
> 
> 
> Hi @matthews-jca, thanks for creating the issue. I assume you mean the verb `echo` instead of `sub`, right?
> 
> This would be a nice thing to add. If you can open a PR for this, feel free to @ me as a reviewer or if you have any questions.

Whoops, you are correct, both verbs for pub and echo are missing those additional QOS settings. 

I have been looking at getting this up as a PR myself as it's been a tricky one for testing from a work perspective. I'll let you know for sure.

NourS-d, 2021-08-29:
I think I can take this. If I understood correctly, it's fairly straight forward.
https://github.com/ros2/ros2cli/blob/master/ros2topic/ros2topic/api/__init__.py#L161
This needs to take into account the liveliness and duration
```    
    if liveliness:
        profile.liveliness = rclpy.qos.QoSLivelinessPolicy.get_from_short_key(liveliness)
    if lease_duration:
        profile.liveliness_lease_duration = lease_duration
```
Then parser args for the pub/echo verbs need to be added and used inside the qos_profile_from_short_keys call.
One question I have though about echo: do we need to reason about the liveliness somehow? https://github.com/ros2/ros2cli/blob/1f4a3aaf7b930830b3fa04773715bf93ea65693e/ros2topic/ros2topic/verb/echo.py#L160

clalancette, 2024-12-27:
We've had this functionality for a while now, so closing this out.

closed 2024-12-27 by clalancette