ros2/ros2cli #619: ros2pkg.api.create_package_environment: handle exports

tim-fan, 2021-04-08:
## Feature request

#### Feature description

In brief, I'm requesting [package.xml.em](https://github.com/ros2/ros2cli/blob/1d1cad3f066a0a43f949d665830bbc772e13d913/ros2pkg/ros2pkg/resource/package_environment/package.xml.em) be updated to allow setting of `export` tags, and [`create_package_environment`](https://github.com/ros2/ros2cli/blob/1d1cad3f066a0a43f949d665830bbc772e13d913/ros2pkg/ros2pkg/api/create.py#L84) be updated to pass `export` tags from the package description to the template.

Background: I'm using `ros2pkg.api` to generate a package.  Looking at [create.py#L148](https://github.com/ros2/ros2cli/blob/1d1cad3f066a0a43f949d665830bbc772e13d913/ros2pkg/ros2pkg/verb/create.py#L148), it looks as if I should be able to list package exports in the `catkin_pkg.package.Package` specification, and have them listed in the `package.xml` generated by [`create_package_environment`](https://github.com/ros2/ros2cli/blob/1d1cad3f066a0a43f949d665830bbc772e13d913/ros2pkg/ros2pkg/api/create.py#L84). However it appears that `create_package_environment` ignores the package export field, and the template only sets the `build_type` export tag.

I found it slightly confusing that  [create.py#L148](https://github.com/ros2/ros2cli/blob/1d1cad3f066a0a43f949d665830bbc772e13d913/ros2pkg/ros2pkg/verb/create.py#L148) sets this export field, even though it is completely ignored by `create_package_environment`

#### Implementation considerations

audrow, 2021-04-16:
Hi @tim-fan, you're right: it is weird that the export field is not used by `create_package_environment`. Feel free to make a pull request for this - you can @ me if you need any help or need a review.

audrow, 2021-05-28:
Closed with #628.

closed 2021-05-28 by audrow