ros2/ros2cli #624: `ros2 topic pub` doesn't publish until after first period has expired

clalancette, 2021-04-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04, arm64
- Installation type:
  - Debian packages
- Version or commit hash:
  - ROS Rolling, package version 0.13.0-1
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

In terminal 1, run:
```
ros2 topic echo /talker std_msgs/msg/String
```

In terminal 2, run:
```
ros2 topic pub -r 0.1 /talker std_msgs/String "data: Hello World"
```
(that is, publish once every 10 seconds)

#### Expected behavior
The `ros2 topic pub` command would publish, then sleep for 10 seconds, then publish, sleep for 10 seconds, etc.

#### Actual behavior
The `ros2 topic pub` command sleeps for 10 seconds, then publishes, then sleeps for 10 seconds, then publishes, etc.

#### Additional information

You could argue that the current way is not wrong, but I think it is unexpected behavior for the user.

fujitatomoya, 2021-04-19:
as far as i know, the current behavior for `once` and `rate` is consistent with ROS1 `rostopic` command. besides, if `rate` is specified by user, probably frequency is what user cares, not starting timepoint.

but one use case that i think of is, if frequency is long enough (like 10 seconds), probably waiting that time window until 1st message would be not really useful.

clalancette, 2021-04-19:
> but one use case that i think of is, if frequency is long enough (like 10 seconds), probably waiting that time window until 1st message would be not really useful.

Yeah, that's exactly where I ran into it.  My example above is a very long time period, and thus the initial message doesn't get published until after the first 10 seconds.

This honestly isn't a huge deal; I doubt too many people are using a 0.1 Hz rate.  I still think it should be fixed, but changing behavior at this point in Galactic is probably a bad idea.  I'll move it to the H-Turtle board instead.

NiklasBurggraaff, 2021-04-29:
I'm looking to get started with contributing to ROS, and I thought this might be a good task to start with. Should I just get started with working on making this change and submit a PR once I'm done?

clalancette, 2021-04-29:
> I'm looking to get started with contributing to ROS, and I thought this might be a good task to start with. Should I just get started with working on making this change and submit a PR once I'm done?

There is already a change proposed to fix this in #626.  One thing to do would be to go ahead and review that change and leave any comments you might have.

If you'd like to look at other tasks to contribute, I'll suggest looking at https://github.com/ros2/ros2cli/issues , and particularly issues with the "help-wanted" tag.  If any of them strike your fancy, it would be a good idea to comment on them and get some feedback before starting on an implementation.

NiklasBurggraaff, 2021-04-29:
> If you'd like to look at other tasks to contribute, I'll suggest looking at https://github.com/ros2/ros2cli/issues , and particularly issues with the "help-wanted" tag. If any of them strike your [...]

Thank you, I will do that

fujitatomoya, 2021-05-12:
@clalancette i will go ahead to close this cz https://github.com/ros2/ros2cli/pull/626 is merged.

clalancette, 2021-05-13:
Yes, we can close this now.  Thank you for the implementation!

closed 2021-05-13 by clalancette