ros2/ros2cli #625: `ros2 param dump` silently ovewrites existing file

clalancette, 2021-04-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Debian packages
- Version or commit hash:
  - ros-rolling-ros2param 0.13.0-1
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

In terminal 1:
```
ros2 run turtlesim turtlesim_node
```

In terminal 2:
```
echo "foo" > turtlesim.yaml  # turtlesim.yaml now contains "foo"
ros2 param dump /turtlesim  # this overwrites the existing turtlesim.yaml, which may be surprising
```

#### Expected behavior

When `ros2 param dump` is choosing a filename, it should either error out if a file with the name it chose already exists, or it should keep choosing names until it finds one that has no conflict.

#### Actual behavior

`ros2 param dump` silently ovewrites the existing file.

hidmic, 2021-04-16:
Sometimes it is practical for it to just overwrite it. Perhaps making opt-in would be best.

clalancette, 2021-04-19:
> Sometimes it is practical for it to just overwrite it. Perhaps making opt-in would be best.

Yeah, I'd be fine with a `--force` or `--overwrite` flag to cause it to always overwrite.  I just think that the current behavior could lead to data loss, where I was storing something I actually wanted in `turtlesim.yaml`, and running `ros2 param dump /turtlesim` in the same directory just overwrites it.

sloretz, 2021-04-19:
What if the default were to write to stdout, with an option to write to a named file? Then overwriting is no big deal because the user had to give a filename to write to.

clalancette, 2021-04-20:
> What if the default were to write to stdout, with an option to write to a named file? Then overwriting is no big deal because the user had to give a filename to write to.

That's a pretty good idea, I like it.  This is also more Unix-y; you could always do `ros2 param dump /turtlesim > turtlesim.yaml`.  

The downside is that this breaks compatibility with earlier releases.  That is, in Foxy, if you do `ros2 param dump /turtlesim`, by default it writes to a file.  If you then pass `--print`, it prints to stdout.  If we were to make this change, then in Galactic `ros2 param dump /turtlesim` would print to stdout, and passing `--write <fname>` (or whatever) would write it to a file.

I don't know, how do people feel about a command-line break like that?

sloretz, 2021-07-12:
@clalancette ok to close now that #638 is merged?

clalancette, 2021-07-13:
Yes!  Closing.

closed 2021-07-13 by clalancette