ros2/ros2cli #645: ros2 param list raises AttributeError

iwmq, 2021-06-02:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04 -->
- Installation type:
  -  Binary installation via apt
- Version or commit hash:
  -  Foxy
- DDS implementation:
  - Default
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
```
ros2 param list
```

#### Expected behavior
List all params without exception
#### Actual behavior
Get exception
```
Exception while calling service of node '/bt_navigator_rclcpp_node': None
Traceback (most recent call last):
  File "/opt/ros/galactic/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.13.1', 'console_scripts', 'ros2')()
  File "/opt/ros/galactic/lib/python3.8/site-packages/ros2cli/cli.py", line 67, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/galactic/lib/python3.8/site-packages/ros2param/command/param.py", line 39, in main
    return extension.main(args=args)
  File "/opt/ros/galactic/lib/python3.8/site-packages/ros2param/verb/list.py", line 116, in main
    sorted_names = sorted(response.result.names)
AttributeError: 'NoneType' object has no attribute 'result'
```
#### Additional information
```
The if statement on `/opt/ros/galactic/lib/python3.8/site-packages/ros2param/verb/list.py line 110` 
checks `future.result()`. If the result is None, it prints a message but allows statements below it
to be executed. Thus cause AttributeError on line 116.

```
----
## Feature request
```
Add a `continue` if  `future.result()` is None.
```
#### Feature description
Allow to skip the None value.

clalancette, 2021-06-02:
That seems like a reasonable change to me.  Please consider contributing a pull request with that change so we can review it and get it in.  While doing that, it would also be awesome to add a test for this case.

iwmq, 2021-06-02:
@clalancette Thanks very much. I'd love to fix this tiny issue. I just created a pull request to the master branch. If you have anything about this process, please tell me.

closed 2021-06-08 by audrow