ros2/ros2cli #648: Can't load non-absolute node name in parameter file

Karsten1987, 2021-06-08:
## Bug report

I have a parameter file in the style of 

```
some_node_name:
  ros__parameters:
    a_list:
        - item1
        - item2
```

#### Expected behavior

`ros2 param load` works to load the correct parameters for node `some_node_name` even though the node name is not specified as absolute.

#### Actual behavior

error is raised: `param file does not contain parameters for /some_node_name,  only for nodes: dict_keys(['some_node_name'])`

Note that there's a simple mismatch between the node names given that one is absolute.

#### Additional information

The reproducer here is based on a ros2-control example in which I would like to use the spawner script to spawn a controller with a given type and a parameter file. 
@v-lopez fyi

v-lopez, 2021-06-08:
I believe this is done on purpose, as the initial behavior allowed both, but after some discussion https://github.com/ros2/ros2cli/pull/590#issuecomment-781506095 and subsequent messages this was the conclusion: https://github.com/ros2/ros2cli/pull/600

v-lopez, 2021-06-08:
I don't know enough about all the use cases, but yeah sounds like a mismatch, which gives more importance to what I requested here https://github.com/ros2/rclpy/issues/671 since basically the same code is reimplemented in multiple places, mismatches like this are bound to happen.

v-lopez, 2021-06-08:
pinging @audrow @ivanpauno @clalancette

ivanpauno, 2021-06-08:
> Isn't there a mismatch in the way we load parameters then?

Maybe, I really don't know why `rcl` accepts a relative name in a parameters file.
Does that only match `/some_node_name` or also `/my_ns/some_node_name`?

I think we can copy what rcl does in `ros2 param load`.
In the case it only matches `/some_node_name`, we could use in the future `/**/some_node_name` to match both options.

fujitatomoya, 2021-07-29:
> Does that only match /some_node_name or also /my_ns/some_node_name?

both match.

i guess `ros2 param load` is the correct behavior. or `rcl_yaml_param_parser`?

fujitatomoya, 2023-10-30:
This has been fixed with https://github.com/ros2/rclpy/commit/3053a8ad9cc266300c7c8e3070bb6552652cf88d

- non-absolute node name case.

```bash
### ros2 param load
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run demo_nodes_cpp parameter_blackboard
[INFO] [1698706220.128919541] [parameter_blackboard]: Parameter blackboard node named '/parameter_blackboard' ready, and serving '6' parameters already!
...
root@tomoyafujita:~/ros2_ws/colcon_ws# cat params.yaml
parameter_blackboard:
  ros__parameters:
    foo: bar
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param load /parameter_blackboard params.yaml
Set parameter foo successful
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param get /parameter_blackboard foo
String value is: bar

### --ros-args --params-file params.yaml
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run demo_nodes_cpp parameter_blackboard --ros-args --params-file params.yaml
[INFO] [1698706337.110460136] [parameter_blackboard]: Parameter blackboard node named '/parameter_blackboard' ready, and serving '7' parameters already!
...
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param get /parameter_blackboard foo
String value is: bar
```

- absolute name case

```bash
### ros2 param load
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run demo_nodes_cpp parameter_blackboard
[INFO] [1698706519.744738775] [parameter_blackboard]: Parameter blackboard node named '/parameter_blackboard' ready, and serving '6' parameters already!
...
root@tomoyafujita:~/ros2_ws/colcon_ws# cat params.yaml
/parameter_blackboard:
  ros__parameters:
    foo: bar
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param load /parameter_blackboard params.yaml
Set parameter foo successful
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param get /parameter_blackboard foo
String value is: bar

### --ros-args --params-file params.yaml
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run demo_nodes_cpp parameter_blackboard --ros-args --params-file params.yaml
[INFO] [1698706478.171368734] [parameter_blackboard]: Parameter blackboard node named '/parameter_blackboard' ready, and serving '7' parameters already!
...
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param get /parameter_blackboard foo
String value is: bar
```

- wildcard case

```bash
### ros2 param load
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run demo_nodes_cpp parameter_blackboard
[INFO] [1698706679.815384266] [parameter_blackboard]: Parameter blackboard node named '/parameter_blackboard' ready, and serving '6' parameters already!
...
root@tomoyafujita:~/ros2_ws/colcon_ws# cat params.yaml
/**/parameter_blackboard:
  ros__parameters:
    foo: bar
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param load /parameter_blackboard params.yaml
Set parameter foo successful
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param get /parameter_blackboard foo
String value is: bar

### --ros-args --params-file params.yaml
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run demo_nodes_cpp parameter_blackboard --ros-args --params-file params.yaml
[INFO] [1698706712.118549727] [parameter_blackboard]: Parameter blackboard node named '/parameter_blackboard' ready, and serving '7' parameters already!
...
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param get /parameter_blackboard foo
String value is: bar
```

closed 2023-10-30 by fujitatomoya