ros2/ros2cli #655: 👨‍🌾 [ros2param] TestVerbList.test_verb_list failure on x86_64 Ubuntu nightly

caguero, 2021-07-16:
Test output:

```
FAIL: test_verb_list.TestVerbList.test_verb_list[rmw_connextdds]
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jenkins-agent/workspace/nightly_linux_debug/ws/src/ros2/ros2cli/ros2param/test/test_verb_list.py", line 172, in test_verb_list
    assert launch_testing.tools.expect_output(
AssertionError: assert False
 +  where False = <function expect_output at 0x7fc861995e50>(expected_lines=['  bool_array_param', '  bool_param', '  double_array_param', '  double_param', '  foo.bar.str_param', '  foo.str_param', ...], text='', strict=True)
 +    where <function expect_output at 0x7fc861995e50> = <module 'launch_testing.tools' from '/home/jenkins-agent/workspace/nightly_linux_debug/ws/install/launch_testing/lib/python3.8/site-packages/launch_testing/tools/__init__.py'>.expect_output
 +      where <module 'launch_testing.tools' from '/home/jenkins-agent/workspace/nightly_linux_debug/ws/install/launch_testing/lib/python3.8/site-packages/launch_testing/tools/__init__.py'> = launch_testing.tools
 +    and   '' = <launch_testing.tools.process.ProcessProxy object at 0x7fc84ff2c940>.output
```

https://ci.ros2.org/view/nightly/job/nightly_linux_debug/1992/

caguero, 2021-07-20:
It might be fixed with https://github.com/eclipse-cyclonedds/cyclonedds/pull/880. Let's confirm with the next job [here](https://build.ros2.org/view/Gci/job/Gci__nightly-release_ubuntu_focal_amd64/).

jacobperron, 2021-07-21:
Locally, I've noticed that `ros2 param list` does not always report a complete list of parameters. I suspect it is because not enough time has been allowed for discovery. To reproduce, run the parameter blackboard demo and try listing it's parameters:

```
RMW_IMPLEMENTATION=rmw_connextdds ros2 run demo_nodes_cpp parameter_blackboard
```

```
RMW_IMPLEMENTATION=rmw_connextdds ros2 param list
```

Repeatedly running the above command, we see that sometimes no parameters are listed.

jacobperron, 2021-07-22:
I think #656 fixed this. The failure didn't appear in any of the latest nightly jobs (though there appears to be another issue related to test shutdown) so I'm closing this out.

closed 2021-07-22 by jacobperron