ros2/ros2cli #657: 👨‍🌾 ros2param tests sometimes do not exit cleanly

jacobperron, 2021-07-22:
## Bug report

**Required Info:**

- Operating System:
  - Any
- Installation type:
  - source
- Version or commit hash:
  - 6cbe07025506233120703bc88e6a2123615e2e97
- DDS implementation:
  - rmw_fastrtps_cpp

#### Steps to reproduce issue

See latest nightly CI jobs, e.g.: 
- https://ci.ros2.org/view/nightly/job/nightly_linux_repeated/2356/testReport/
- https://ci.ros2.org/view/nightly/job/nightly_linux-aarch64_repeated/1663/testReport/

The tests appear flaky as I've only seen the failures on the repeated jobs.

Test results:

```
FAIL: test_verb_load.TestVerbDump.test_verb_load[rmw_fastrtps_cpp]
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/src/ros2/ros2cli/ros2param/test/test_verb_load.py", line 279, in test_verb_load
    assert param_load_command.exit_code == launch_testing.asserts.EXIT_OK
AssertionError: assert 1 == 0
 +  where 1 = <launch_testing.tools.process.ProcessProxy object at 0xffffa610f790>.exit_code
 +  and   0 = <module 'launch_testing.asserts' from '/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/install/launch_testing/lib/python3.8/site-packages/launch_testing/asserts/__init__.py'>.EXIT_OK
 +    where <module 'launch_testing.asserts' from '/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/install/launch_testing/lib/python3.8/site-packages/launch_testing/asserts/__init__.py'> = launch_testing.asserts

===================================================================================================================================================================================================================================
FAIL: test_verb_load.TestVerbDump.test_verb_load_wildcard[rmw_fastrtps_cpp]
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/src/ros2/ros2cli/ros2param/test/test_verb_load.py", line 318, in test_verb_load_wildcard
    assert param_load_command.exit_code == launch_testing.asserts.EXIT_OK
AssertionError: assert 1 == 0
 +  where 1 = <launch_testing.tools.process.ProcessProxy object at 0xffffa60a6f40>.exit_code
 +  and   0 = <module 'launch_testing.asserts' from '/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/install/launch_testing/lib/python3.8/site-packages/launch_testing/asserts/__init__.py'>.EXIT_OK
 +    where <module 'launch_testing.asserts' from '/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/install/launch_testing/lib/python3.8/site-packages/launch_testing/asserts/__init__.py'> = launch_testing.asserts
 ```

```
FAIL: test_verb_list.TestVerbList.test_verb_list[rmw_fastrtps_cpp]
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/install/launch_testing/lib/python3.8/site-packages/launch_testing/markers.py", line 57, in _wrapper
    return func(self, *args, **kwargs)
  File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/src/ros2/ros2cli/ros2param/test/test_verb_list.py", line 171, in test_verb_list
    assert param_list_command.wait_for_shutdown(timeout=TEST_TIMEOUT)
AssertionError: assert False
 +  where False = <bound method ProcessProxy.wait_for_shutdown of <launch_testing.tools.process.ProcessProxy object at 0xffffa61c1790>>(timeout=20.0)
 +    where <bound method ProcessProxy.wait_for_shutdown of <launch_testing.tools.process.ProcessProxy object at 0xffffa61c1790>> = <launch_testing.tools.process.ProcessProxy object at 0xffffa61c1790>.wait_for_shutdown

==========================================================================================================================================================================================================================
FAIL: test_verb_list.TestVerbList.test_verb_list_filter[rmw_fastrtps_cpp]
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/install/launch_testing/lib/python3.8/site-packages/launch_testing/markers.py", line 57, in _wrapper
    return func(self, *args, **kwargs)
  File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/src/ros2/ros2cli/ros2param/test/test_verb_list.py", line 195, in test_verb_list_filter
    assert param_list_command.wait_for_shutdown(timeout=TEST_TIMEOUT)
AssertionError: assert False
 +  where False = <bound method ProcessProxy.wait_for_shutdown of <launch_testing.tools.process.ProcessProxy object at 0xffffa6162fd0>>(timeout=20.0)
 +    where <bound method ProcessProxy.wait_for_shutdown of <launch_testing.tools.process.ProcessProxy object at 0xffffa6162fd0>> = <launch_testing.tools.process.ProcessProxy object at 0xffffa6162fd0>.wait_for_shutdown
```

```
FAIL: test_verb_list.TestVerbList.test_verb_list[rmw_fastrtps_cpp]
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/install/launch_testing/lib/python3.8/site-packages/launch_testing/markers.py", line 57, in _wrapper
    return func(self, *args, **kwargs)
  File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/src/ros2/ros2cli/ros2param/test/test_verb_list.py", line 171, in test_verb_list
    assert param_list_command.wait_for_shutdown(timeout=TEST_TIMEOUT)
AssertionError: assert False
 +  where False = <bound method ProcessProxy.wait_for_shutdown of <launch_testing.tools.process.ProcessProxy object at 0xffffa61c1790>>(timeout=20.0)
 +    where <bound method ProcessProxy.wait_for_shutdown of <launch_testing.tools.process.ProcessProxy object at 0xffffa61c1790>> = <launch_testing.tools.process.ProcessProxy object at 0xffffa61c1790>.wait_for_shutdown

==========================================================================================================================================================================================================================
FAIL: test_verb_list.TestVerbList.test_verb_list_filter[rmw_fastrtps_cpp]
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/install/launch_testing/lib/python3.8/site-packages/launch_testing/markers.py", line 57, in _wrapper
    return func(self, *args, **kwargs)
  File "/home/jenkins-agent/workspace/nightly_linux-aarch64_repeated/ws/src/ros2/ros2cli/ros2param/test/test_verb_list.py", line 195, in test_verb_list_filter
    assert param_list_command.wait_for_shutdown(timeout=TEST_TIMEOUT)
AssertionError: assert False
 +  where False = <bound method ProcessProxy.wait_for_shutdown of <launch_testing.tools.process.ProcessProxy object at 0xffffa6162fd0>>(timeout=20.0)
 +    where <bound method ProcessProxy.wait_for_shutdown of <launch_testing.tools.process.ProcessProxy object at 0xffffa6162fd0>> = <launch_testing.tools.process.ProcessProxy object at 0xffffa6162fd0>.wait_for_shutdown
```

jacobperron, 2021-07-22:
Possibly a duplicate of https://github.com/ros2/rmw_fastrtps/issues/531

clalancette, 2024-12-27:
I don't think we've seen this for a while, so closing this out.  Feel free to reopen if you notice it again.

closed 2024-12-27 by clalancette