ros2/ros2cli #662: ros2 param list doesn't work while rqt is running

Kaju-Bubanja, 2021-08-24:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - ros2 master source checkout
- Version or commit hash:
  - 1f4a3aaf7b930830b3fa04773715bf93ea65693e
- DDS implementation:
  - rmw_fastrtps_cpp
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
When running `rqt` and then `ros2 param list` I get 
`Exception while calling service of node '/rqt_gui_py_node_6150': None`
#### Expected behavior
I would expect it to tell me what params there are
#### Actual behavior
I get following exception: `Exception while calling service of node '/rqt_gui_py_node_6150': None`
#### Additional information
The same happens for nodes that I have written, but it also happens after a complete reboot and just starting `rqt` and running `ros2 param list`. I confirmed that no other node is running and messing something up with `ros2 node list` which only reports `/rqt_gui_py_node_6150`

clalancette, 2021-08-30:
Same here; I cannot reproduce this issue on the latest ros2 code.

@Kaju-Bubanja Can you explain further about how you have rqt configured?  Maybe there is some plugin that we need loaded in order to see this issue.

Kaju-Bubanja, 2021-08-30:
I will try to reproduce this week with the hash of ros2 you tried.

Kaju-Bubanja, 2021-08-31:
Could not reproduce either.

closed 2021-08-31 by Kaju-Bubanja