ros2/ros2cli #667: Topic Info verbose doesn't show pub/sub queue depth

matthews-jca, 2021-09-29:
## Feature request

#### Feature description
when running ros2 topic info with the --verbose nothing is exposed for the publisher/subscriber queue depths (when using keep last N)

Not sure if it's as simple as adding in the extra parameter https://github.com/ros2/rclpy/blob/691e4fbfcb4bd4cc2a01182a7dced3105a78200b/rclpy/rclpy/topic_endpoint_info.py#L32 or if this needs to be exposed at the C level first.

matthews-jca, 2021-10-04:
As this is part of the qosprofile itself I believe the actual fix is just to add the qosprofile.depth https://github.com/ros2/rclpy/blob/691e4fbfcb4bd4cc2a01182a7dced3105a78200b/rclpy/rclpy/topic_endpoint_info.py#L169 correct? If so, I can get a PR up at the rclpy level.

fujitatomoya, 2021-11-09:
@matthews-jca yes, i think that will do the all the trick.

fujitatomoya, 2021-11-13:
@matthews-jca i just created https://github.com/ros2/rclpy/pull/849, it will be nice if you could help review on that :bow:

matthews-jca, 2021-11-17:
Following up from https://github.com/ros2/rclpy/pull/849#issuecomment-970912206 being not shared across all RMWs, it's still super helpful and useful to get those settings when available. I know hand coding a per-rmw solution is unsustainable, but would a separate cli command be better here?

clalancette, 2024-12-27:
This was fixed a long time ago, so closing this out.

closed 2024-12-27 by clalancette