ros2/ros2cli #671: Add "debug" command

roncapat, 2021-11-10:
The `debug` command would be like the `run` or `launch` command, but instead of running the node directly, it runs inside `gdb` or any configured debugger. 

Currently, whenever a node fails, I copy the full command line that is printed in the last line of logs, type `gdb --args ` + the copied command line, and debug the node. But a proper ROS2 CLI tool would be awesome.

sisaha9, 2023-07-07:
@roncapat may want to look at https://navigation.ros.org/tutorials/docs/get_backtrace.html for some ideas on debugging. They should be easier than what you are doing right now. In particular the `backward_ros` has been very helpful

clalancette, 2023-07-07:
I have a couple of thoughts about this.

First, I don't really think this would make sense as part of a `ros2 launch` command.  Launch is typically used to launch multiple processes, and so it is unclear which one you should attach to.  There are facilities in launch for adding a prefix to a specific process (using the `prefix`), so I think that is what should be used there.

Second, `ros2 run` already has the command-line argument `--prefix`, which allows you to specify an arbitrary prefix when running something.  That is, you can do `ros2 run --prefix gdb demo_nodes_cpp talker`, and it will launch `talker` under gdb.

Thus, I think most of this functionality already exists.  What we are probably missing is documentation in the core about it, so if you'd like to improve that, please consider opening a pull request to https://github.com/ros2/ros2_documentation.

roncapat, 2023-07-07:
Thank you both!
You are right, this is the kind of things that should be made clearer by means of docs. I will try to find the time / proper doc place to do it :)