ros2/ros2cli #675: newly created package fails colcon test

christianrauch, 2021-11-28:
## Bug report

Creating a new node with ros2 creates a package that fails colcon test.

**Required Info:**

- Operating System: Ubuntu 20.04
- Installation type: binaries
- Version or commit hash: 0.9.10

#### Steps to reproduce issue
1. create a new node: `ros2 pkg create my_package --build-type ament_cmake --dependencies rclcpp --node-name my_node`
2. run tests: `colcon build` , `colcon test`

#### Expected behavior

A newly created package should follow ROS2 standards and not fail its own tests.

#### Actual behavior

`colcon test` fails

clalancette, 2021-11-28:
So looking at the failures, they both revolve around the copyright.  This makes some sense, as the copyright wasn't specified in the initial `ros2 pkg create`.  That said, adding the `--license` flag doesn't improve the situation, because the copyright lines are not added to the top of the skeleton main.

Unfortunately, we don't have all of the information required in order to properly fill in that copyright header.  In order to do that, we need both the copyright header text and the copyright holder.  If the user specifies `--license`, and it is one of the licenses we know, then we know the copyright header text.  But we currently have no way to get the copyright holder.

At the end of the day, this is a tradeoff.  If we want to be rigorous and ensure that we'll pass the tests, then we'd need to have a required flag for the license (which we already have), and a required flag for the copyright holder (which we would have to add).  This would go further towards our goal of making all packages have some license, at the expense of making this utility more pedantic and more difficult to use.  The other way you could argue this is that the license is something that we want the user to examine and make deliberate choices about, so having it fail the tests out-of-the-box is the way it should be.

I'm on the fence about which way to go here.  @ros2/team any opinions?

christianrauch, 2021-11-28:
> But we currently have no way to get the copyright holder.

What about just using the `MAINTAINER_NAME`? When you create the package, this is most likely the person that is going to write the initial code.

> required flag for the copyright holder

In case no copyright holder is specified or it is not taken from the maintainer, I would just deactivate the test in the CMake (`set(ament_cmake_copyright_FOUND TRUE)`).

> This would go further towards our goal of making all packages have some license

I wouldn't mind if package creators are forced to choose a licence when publishing code online. This makes it much easier to reuse packages. But this makes it also a bit cumbersome for users that just want to create a package for personal use.

In summary, I think that those values should either be determined automatically from the provided parameters, or the tests should be deactivated if required values cannot be determined.

aprotyas, 2021-11-28:
One way to address the copyright issue is to disable the copyright test by default (comment out line 82 in the linked snippet), allowing the package author to enable it later just by commenting a single line in the CMakeLists.txt file.

https://github.com/ros2/ros2cli/blob/37fea0685b9b4ac0092d355d455ce3e53c29858c/ros2pkg/ros2pkg/resource/ament_cmake/CMakeLists.txt.em#L80-L82

That would still keep the tool relatively easy to use.

For "smart" default behavior, maybe the `set(ament_cmake_copyright_FOUND TRUE)` can be conditionally commented out in the CMakeLists.txt template if the user specifies a valid/known `--license` option?

christianrauch, 2021-11-28:
I noticed that even after disabling `ament_cmake_copyright` by commenting that line, `colcon test-result --verbose` will still show the issue. It's just that `colcon test` will not fail.

clalancette, 2021-11-29:
> > 
> 
> What about just using the `MAINTAINER_NAME`? When you create the package, this is most likely the person that is going to write the initial code.

We could do that, but I think there are no instances in the core codebase where the maintainer name is the same as the copyright holder name.  That is, the maintainer is generally a person, while the copyright holder is generally an organization.  So while we could rely on this, it will probably be wrong (and will pass the tests, meaning that the developer may overlook changing it).

> One way to address the copyright issue is to disable the copyright test by default (uncomment line 82 in the linked snippet), allowing the package author to enable it later just by commenting a single line in the CMakeLists.txt file.

I would be onboard with that.  It doesn't fail the tests out-of-the-box, but it is at least a marker that the developer should do something about it.

jacobperron, 2021-11-30:
> I noticed that even after disabling ament_cmake_copyright by commenting that line, colcon test-result --verbose will still show the issue. It's just that colcon test will not fail.

@christianrauch This could be due to a left-over test result. Try again in a clean workspace (e.g. remove the `build`, `install` and `log` directories).

---

+1 for disabling the test by default.

We can still pester the user with comments in the code to encourage them to add a license/test.

closed 2021-11-30 by audrow