ros2/ros2cli #677: ros2 param set string type "Off"

icolwell-as, 2021-12-07:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - Foxy
- DDS implementation:
  - Default
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Declare string parameter from within cpp code.
Attempt to set the parameter from the CLI to the string "Off"
```
ros2 param set /node param_name Off
```

#### Expected behavior

Paremeter is set to the string "Off"

#### Actual behavior

Parameter is changed to boolean type and set to False.

#### Additional information

I tried all the following, none of them allow me to set a string value:
```
ros2 param set /node param_name Off
ros2 param set /node param_name "Off"
ros2 param set /node param_name 'Off'
ros2 param set /node param_name off
```

Is there any way to enforce the type when parsing the passed-in CLI value? or at the very least assume the passed-in value is the same type as the existing parameter type?

Maybe a new `-f`/`--force-type` flag for "force" type?

clalancette, 2021-12-23:
Thanks for the report.  This is basically unintentional behavior (and hence a bug).

What's happening here is that we are using YAML parsing to parse the value, and it is automatically interpreting certain values (on, off, true, false, etc) as booleans, and then trying to send that to the node.  In Foxy, parameter types are flexible, so it automatically changes the type.  In Rolling, for instance, the `ros2 param set` call fails because there we are enforcing types and this is trying to change the type.

One workaround for all platforms is to do the following:

```
ros2 param set /node param_name "\"off\""
```

Which will make the parameter literally `"off"`.  That's enough to keep it a string, though you'll have to have custom code inside of the node in order to strip the quotes off.  In the meantime, I'll mark this as a bug to look into.

clalancette, 2021-12-23:
https://github.com/ros2/ros2cli/pull/684 is a potential fix for this, though I don't think we can completely fix it.

With #684 in, running:

```
ros2 param set /node param 'off'
```

will still interpret the argument as a boolean.  But the following would work as an override:

```
ros2 param set /node param '!!str off'
```

closed 2022-01-06 by clalancette

icolwell-as, 2022-01-06:
Thanks for the fix! much appreciated.

clalancette, 2022-01-06:
You are welcome, happy to help.

PR #684 (2022-01-06): Fix how ros2 param interprets command-line arguments.
Basically ensure that we are *always* using the parsed yaml
string, so that users can override types as appropriate.
This makes it possible for users to specify something like:

ros2 param set /node param '!!str off'

for force a string to be "off" as opposed to having YAML
intepret it as a boolean.

This is still a draft because I'm kind of nervous about the change; this has the potential to break some use-cases that worked before.  All of the tests pass, and basic testing here has worked fine for me, but I want to do more testing before m [... cut]