ros2/ros2cli #679: 'property' object is not callable

fujitatomoya, 2021-12-13:
## Bug report

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - source build
- Version or commit hash:
  - https://github.com/ros2/ros2/commit/4a36f319afcc208deae3c373493ef6d9feda1324
- DDS implementation:
  - anything (cyclonedds, Fast-DDS, ConnextDDS)
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

the following colcon workspace includes reproducible package.

```bash
### how to build
cd <your colcon workspace>/src/ros2
git clone git@github.com:fujitatomoya/ros2_test_prover.git
cd <your colcon workspace>
colcon build --symlink-install --packages-select prover_rclcpp prover_rclpy prover_interfaces
source install/local_setup.bash

### terminal-1
ros2 run prover_rclcpp ros2cli_679
...

### terminal-2
ros2 topic hz /property
Traceback (most recent call last):
  File "/root/ros2_ws/master_ws/install/ros2cli/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli', 'console_scripts', 'ros2')())
  File "/root/ros2_ws/master_ws/build/ros2cli/ros2cli/cli.py", line 89, in main
    rc = extension.main(parser=parser, args=args)
  File "/root/ros2_ws/master_ws/build/ros2topic/ros2topic/command/topic.py", line 41, in main
    return extension.main(args=args)
  File "/root/ros2_ws/master_ws/build/ros2topic/ros2topic/verb/hz.py", line 87, in main
    return main(args)
  File "/root/ros2_ws/master_ws/build/ros2topic/ros2topic/verb/hz.py", line 102, in main
    _rostopic_hz(node.node, topic, window_size=args.window_size, filter_expr=filter_expr,
  File "/root/ros2_ws/master_ws/build/ros2topic/ros2topic/verb/hz.py", line 256, in _rostopic_hz
    msg_class = get_msg_class(
  File "/root/ros2_ws/master_ws/build/ros2topic/ros2topic/api/__init__.py", line 98, in get_msg_class
    msg_class = _get_msg_class(node, topic, include_hidden_topics)
  File "/root/ros2_ws/master_ws/build/ros2topic/ros2topic/api/__init__.py", line 144, in _get_msg_class
    return get_message(message_type)
  File "/root/ros2_ws/master_ws/build/rosidl_runtime_py/rosidl_runtime_py/utilities.py", line 28, in get_message
    interface = import_message_from_namespaced_type(get_message_namespaced_type(identifier))
  File "/root/ros2_ws/master_ws/build/rosidl_runtime_py/rosidl_runtime_py/import_message.py", line 30, in import_message_from_namespaced_type
    module = importlib.import_module(
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/root/ros2_ws/master_ws/install/prover_interfaces/lib/python3.8/site-packages/prover_interfaces/msg/__init__.py", line 1, in <module>
    from prover_interfaces.msg._property import Property  # noqa: F401
  File "/root/ros2_ws/master_ws/install/prover_interfaces/lib/python3.8/site-packages/prover_interfaces/msg/_property.py", line 52, in <module>
    class Property(metaclass=Metaclass_Property):
  File "/root/ros2_ws/master_ws/install/prover_interfaces/lib/python3.8/site-packages/prover_interfaces/msg/_property.py", line 131, in Property
    def anything(self):
TypeError: 'property' object is not callable
```

#### Expected behavior

`ros2 topic` command issued successfully.

#### Actual behavior

failed with `'property' object is not callable`

#### Additional information

This problem only happens if message type is defined after `property` message type.

https://github.com/fujitatomoya/ros2_test_prover/blob/master/prover_interfaces/msg/Property.msg

```
string property
string anything
```

fujitatomoya, 2021-12-13:
CC: @iuhilnehc-ynos @Barry-Xu-2018

fujitatomoya, 2021-12-13:
A003 ; 'xxx' is a python builtin, consider renaming the class attribute. (https://pypi.org/project/flake8-awesome/)

so this is probably not a bug?

Barry-Xu-2018, 2021-12-14:
> A003 ; 'xxx' is a python builtin, consider renaming the class attribute. (https://pypi.org/project/flake8-awesome/)
>
> so this is probably not a bug?

Agree. It isn't a real bug.  
While `property` name is used in msg, generated python code has this problem.   
Maybe we should change code which generate python code from msg to report error if `property` name is used in msg (Not sure if generated code for srv/action has this issue. Need to check).  

Use below python code to reproduce this problem 
```python
class test:

    __slot__ = ['_property', '_anything']

    @property
    def property(self):
        print("Getting property ...")
        return self._property

    @property.setter
    def property(self, value):
        print("Setting property...")
        self._property = value
      
    @property
    def anything(self):
        print("Getting anything ...")
        return self._anything
    
    @anything.setter
    def anything(self, value):
        print("Setting property...")
        self._anything = value

test1 = test()

test1.property = 100
test1.anything = 110

print(test1.property)
print(test1.anything)
```
After execute, report error
```
raceback (most recent call last):
  File "./test.py", line 4, in <module>
    class test:
  File "./test.py", line 19, in test
    def anything(self):
TypeError: 'property' object is not callable
```

iuhilnehc-ynos, 2021-12-14:
@Barry-Xu-2018 @fujitatomoya 

I think that many users hope it could work. A workaround is making the `property` field generated at the end. (Need to update the https://github.com/ros2/rosidl_python.git)

Barry-Xu-2018, 2021-12-14:
> I think that many users hope it could work. A workaround is making the property field generated at the end. (Need to update the https://github.com/ros2/rosidl_python.git, but maintainers might not accept it.)

Yes. This is a good idea.  But maybe it makes the logic of rosidl_python more complex.  
We should hear the maintainer's opinion.   

@fujitatomoya It is better to move this issue to [rosidl_python](https://github.com/ros2/rosidl_python/issues) for further discussion.

clalancette, 2021-12-14:
One thing we have discussed in the past is to change the message generator to always add an `_` to the end of every field in the message.  This would ensure that we never run into these type of problems, though the downside is that we would break every Python application that is currently using the fields.  Maybe we could do a deprecation cycle, though.

fujitatomoya, 2021-12-14:
@clalancette thanks for the comment.

i think we are okay with either way.
- add documentation that `property` (python buildin) message field name could lead this error, so user should not use that.
- address this with `_` to make it work. (or workaround suggested by @iuhilnehc-ynos  https://github.com/ros2/ros2cli/issues/679#issuecomment-993260445)

i like to leave this to maintainer.

btw, can you move this issue to https://github.com/ros2/rosidl_python/issues?