ros2/ros2cli #680: ros2 topic echo /odom covariance is

RENyunfan, 2021-12-16:
## Bug report

**Required Info:**

- Operating System: Ubuntu20.04
  - 
- Installation type: binaries
  - 
- Version or commit hash: ROS2 Foxy
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
ros2 topic echo /odom
```

#### Expected behavior
The covariance should line up so we can see it better. But they're actually arranged vertically.
#### Actual behavior
```bash
header:
  stamp:
    sec: 1
    nanosec: 639623972
  frame_id: /world
child_frame_id: /quadrotor
pose:
  pose:
    position:
      x: 0.0
      y: 0.0
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
  covariance:
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
twist:
  twist:
    linear:
      x: 0.0
      y: 0.0
      z: 0.0
    angular:
      x: 0.0
      y: 0.0
      z: 0.0
  covariance:
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
---

```
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

clalancette, 2021-12-16:
This is actually difficult to do, as `ros2 topic echo` doesn't have any knowledge of the data it is printing.  It doesn't know if the given array should be arranged vertically, horizontally, or with some stride.

I'll leave this open as a feature request, because it would be useful, but someone would have to come up with some concrete proposals on how we would achieve this.

fujitatomoya, 2021-12-16:
this sounds reasonable to me 👍 

```bash
### BEFORE
orientation_covariance:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0

### AFTER
orientation_covariance: ['0.0', '0.0', '0.0', '0.0', '0.0', '0.0', '0.0', '0.0', '0.0']
```

https://github.com/ros2/rosidl_runtime_py/blob/f1bfe509c457e4693f8c1e85a5cecdd8483bd7d2/rosidl_runtime_py/convert.py#L66

probably we would want to add format option, and then we can issue it via ros2cli?

iuhilnehc-ynos, 2021-12-17:
> vertically, horizontally

Just adding the `default_flow_style=True` to `yaml.dump` could work.

> or with some stride

I tried to add `yaml.add_representer(list, __represent_list)`, but it doesn't work.
After trying to add a new `Ros2Dumper` derived from `yaml.Dumper`, and overwrite the `expect_node` method, it seems to work as expected.
Using `stride = 8` as following,

```sh
child_frame_id: base_footprint
pose:
  pose:
    position:
      x: -1.9890866547852504
      y: -0.4965558336701039
      z: 0.008716560937518051
    orientation:
      x: -0.0002477036005055618
      y: 0.000739928503462637
      z: 0.3268246241420463
      w: 0.9450846820275663
  covariance: [
      1.0e-05, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0e-05,
      0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1000000000000.0, 0.0,
      0.0, 0.0, 0.0, 0.0, 0.0, 1000000000000.0, 0.0, 0.0,
      0.0, 0.0, 0.0, 0.0, 1000000000000.0, 0.0, 0.0, 0.0,
      0.0, 0.0, 0.0, 0.001]
twist:
  twist:
    linear:
      x: 6.897880654430438e-05
      y: 1.591879781315907e-05
      z: 0.0
    angular:
      x: 0.0
      y: 0.0
      z: -0.0002861788819025045
  covariance: [
      1.0e-05, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0e-05,
      0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1000000000000.0, 0.0,
      0.0, 0.0, 0.0, 0.0, 0.0, 1000000000000.0, 0.0, 0.0,
      0.0, 0.0, 0.0, 0.0, 1000000000000.0, 0.0, 0.0, 0.0,
      0.0, 0.0, 0.0, 0.001]

```

So I think we can provide two extra options to users,
```sh
'--show-list-pretty' 
    True : it shows [item, item2, ..., itemn]
    False : it shows
         - item
         - item2
         - ...
         - itemn
'--list-stride':
   Number: effective only if '--show-list-pretty' is True.
```

iuhilnehc-ynos, 2021-12-17:
It seems a little complicated in some situations (embedded lists).

RENyunfan, 2021-12-17:
> > vertically, horizontally
> 
> Just adding the `default_flow_style=True` to `yaml.dump` could work.
> 
> > or with some stride
> 
> I tried to add `yaml.add_representer(list, __represent_list)`, but it doesn't work. After trying to add a new `Ros2Dumper` derived from `yaml.Dumper`, and overwrite the `expect_node` method (unmaintainable), it seems to work as expected. Using `stride = 8` as following,
> 
> ```shell
> child_frame_id: base_footprint
> pose:
>   pose:
>     position:
>       x: -1.987142582426784
>       y: -0.4948242098096342
>       z: 0.008726311344861208
>     orientation:
>       x: -0.00029216596084799025
>       y: 0.0007255788333816825
>       z: 0.3837788667705221
>       w: 0.9234246962231094
>   covariance: [
>     1.0e-05, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0e-05,
>     0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1000000000000.0, 0.0,
>     0.0, 0.0, 0.0, 0.0, 0.0, 1000000000000.0, 0.0, 0.0,
>     0.0, 0.0, 0.0, 0.0, 1000000000000.0, 0.0, 0.0, 0.0,
>     0.0, 0.0, 0.0, 0.001]
> twist:
>   twist:
>     linear:
>       x: 6.673055220339329e-05
>       y: 1.066510747665083e-05
>       z: 0.0
>     angular:
>       x: 0.0
>       y: 0.0
>       z: -0.00030110521608306537
>   covariance: [
>     1.0e-05, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0e-05,
>     0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1000000000000.0, 0.0,
>     0.0, 0.0, 0.0, 0.0, 0.0, 1000000000000.0, 0.0, 0.0,
>     0.0, 0.0, 0.0, 0.0, 1000000000000.0, 0.0, 0.0, 0.0,
>     0.0, 0.0, 0.0, 0.001]
> 
> (Updated: make sure no more extra indent)
> ```
> 
> So I think we can provide two extra options to users,
> 
> ```shell
> '--show-list-pretty' 
>     True : it shows [item, item2, ..., itemn]
>     False : it shows
>          - item
>          - item2
>          - ...
>          - itemn
> '--list-stride':
>    Number: effective only if '--show-list-pretty' is True.
> ```

Hi, thx for your reply, I tried to modify `default_flow_style` in `/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py ` and `/opt/ros/foxy/lib/python3.8/site-packages/ros2param/verb/dump.py`, but nothing happened. Is there any detail for your solution?

iuhilnehc-ynos, 2021-12-17:
Please refer to https://github.com/ros2/ros2cli/issues/680#issuecomment-996227586

RENyunfan, 2021-12-17:
> Please refer to [#680 (comment)](https://github.com/ros2/ros2cli/issues/680#issuecomment-996227586)
Thanks, I'd like to post the detailed steps below for anyone who has the same problem.
In `/opt/ros/foxy/lib/python3.8/site-packages/rosidl_runtime_py/convert.py`, line 66-94, add a param in line 93 will work.
```python
def message_to_yaml(
    msg: Any,
    *,
    truncate_length: int = None,
    no_arr: bool = False,
    no_str: bool = False
) -> str:
    """
    Convert a ROS message to a YAML string.

    :param msg: The ROS message to convert.
    :param truncate_length: Truncate values for all message fields to this length.
        This does not truncate the list of message fields.
    :param no_arr: Exclude array fields of the message.
    :param no_str: Exclude string fields of the message.
    :returns: A YAML string representation of the input ROS message.
    """
    global __yaml_representer_registered

    # Register our custom representer for YAML output
    if not __yaml_representer_registered:
        yaml.add_representer(OrderedDict, __represent_ordereddict)
        __yaml_representer_registered = True

    return yaml.dump( 
        message_to_ordereddict(
            msg, truncate_length=truncate_length, no_arr=no_arr, no_str=no_str),
        allow_unicode=True, width=sys.maxsize, default_flow_style=True,
    )
```

closed 2022-03-17 by fujitatomoya