ros2/ros2cli #699: Cannot change log level on ros2 component load command (galactic)

Nir-Az, 2022-03-07:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 20.04
- Installation type:
binaries
- Version or commit hash:
galactic
- DDS implementation:
default

#### Steps to reproduce issue
Try to change log level on `ros2 component load` command

#### Expected behavior
Node loaded with requested log level
#### Actual behavior
Error from the command
#### Additional information

$ ros2 component load /ComponentManager camera camera::NodeFactory --log-level  WARN
Traceback (most recent call last):
  File "/opt/ros/galactic/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.13.1', 'console_scripts', 'ros2')()
  File "/opt/ros/galactic/lib/python3.8/site-packages/ros2cli/cli.py", line 67, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/galactic/lib/python3.8/site-packages/ros2component/command/component.py", line 37, in main
    return extension.main(args=args)
  File "/opt/ros/galactic/lib/python3.8/site-packages/ros2component/verb/load.py", line 49, in main
    component_uid, component_name = load_component_into_container(
  File "/opt/ros/galactic/lib/python3.8/site-packages/ros2component/api/__init__.py", line 257, in load_component_into_container
    request.log_level = log_level
  File "/opt/ros/galactic/lib/python3.8/site-packages/composition_interfaces/srv/_load_node.py", line 217, in log_level
    assert \
AssertionError: The 'log_level' field must be of type 'int'

fujitatomoya, 2022-03-07:
according to the following experience, we would need to support

- `--ros-args` option.
- individual log level for each node.

any comments and thoughts?

```
# ros2 run rclcpp_components component_container --ros-args --log-level debug
```

this works but all nodes will set the same log level with container.

```
# ros2 component load /ComponentManager composition composition::Talker --ros-args --log-level debug
usage: ros2 [-h] [--use-python-default-buffering] Call `ros2 <command> -h` for more detailed usage. ...
ros2: error: unrecognized arguments: --ros-args
```

expected this would work, but it does not support `--ros-args` option.

```
# ros2 component load /ComponentManager composition composition::Talker --log-level warn
Traceback (most recent call last):
  File "/root/ros2_ws/colcon_ws/install/ros2cli/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli', 'console_scripts', 'ros2')())
  File "/root/ros2_ws/colcon_ws/build/ros2cli/ros2cli/cli.py", line 89, in main
    rc = extension.main(parser=parser, args=args)
  File "/root/ros2_ws/colcon_ws/build/ros2component/ros2component/command/component.py", line 37, in main
    return extension.main(args=args)
  File "/root/ros2_ws/colcon_ws/build/ros2component/ros2component/verb/load.py", line 49, in main
    component_uid, component_name = load_component_into_container(
  File "/root/ros2_ws/colcon_ws/build/ros2component/ros2component/api/__init__.py", line 229, in load_component_into_container
    request.log_level = log_level
  File "/root/ros2_ws/colcon_ws/install/composition_interfaces/lib/python3.8/site-packages/composition_interfaces/srv/_load_node.py", line 219, in log_level
    assert \
AssertionError: The 'log_level' field must be of type 'int'
```

problem can be observed in mainline.

Nir-Az, 2022-03-08:
I think the best way to go is to align to the `ros2 run command` syntax.

I wouldn't open it as a BUG if it wasn't listed in the command help syntax.. I spend 2 hours trying to understand what am I doing wrong, because if it listed in the help it should work :)

```
ros2 component load --help
usage: ros2 component load [-h] [--spin-time SPIN_TIME] [-s] [--no-daemon]
                           [-n NODE_NAME] [--node-namespace NODE_NAMESPACE]
                           [--log-level LOG_LEVEL] [-r REMAP_RULES]
                           [-p PARAMETERS] [-e EXTRA_ARGUMENTS] [-q]
                           container_node_name package_name plugin_name
```

JGyimah, 2026-10-16:
This seems to be fixed on `rolling` but still present on every maintained distro.

The fix was #1212 ("add `--log-level` option for `ros2 component load`", 3e1c2ea41c, 2026-05-29), which converts the string before assigning it:

```python
severity = get_logging_severity_from_string(log_level)
request.log_level = int(severity)
```

and restricts the argument with `choices=['debug', 'info', 'warn', 'error', 'fatal']`.

On `humble`, `jazzy` and `kilted`, `ros2component/api/__init__.py:229` still has the original line:

```python
request.log_level = log_level
```

`log_level` there is whatever argparse produced, a `str`, and `LoadNode.log_level` is a `uint8`, so the request cannot be converted. On Iron the failure is not the traceback in the original report but an abort, since the generated conversion code asserts rather than raising:

```
$ ros2 run rclcpp_components component_container &
$ ros2 component load /ComponentManager rclcpp_components anything --log-level 10
python3: ./.obj-x86_64-linux-gnu/rosidl_generator_py/composition_interfaces/srv/_load_node_s.c:135: composition_interfaces__srv__load_node__request__convert_from_py: Assertion `PyLong_Check(field)' failed.
```

The same command without `--log-level` returns a normal error, so the flag is what fails. Numeric values abort too: argparse has no `type`, so `10` is also a `str`. There is no value of `--log-level` that works.

That is the stock C++ container, so this is not specific to a client library.

Would you consider backporting #1212 to jazzy and humble? The field has been unusable since this issue was opened in 2022, and on binaries the symptom is a SIGABRT rather than a Python error, which makes it look like a crash in the container rather than a CLI argument problem.

Tested on Iron binaries, ros2cli 0.25.8, Ubuntu 22.04, Python 3.10.12.

fujitatomoya, 2026-10-19:
> Would you consider backporting https://github.com/ros2/ros2cli/pull/1212 to jazzy and humble?

for doing that, 1st we need ABI-safe variant of https://github.com/ros2/rclcpp/pull/3092. i would probably not do that by myself.

JGyimah, 2026-10-20:
Thanks, that makes sense. I'll leave the ABI-safe variant of ros2/rclcpp#3092 to someone with more rclcpp context than me — adding to `NodeOptions` in a released distro isn't something I'd want to get wrong.

Having read #3092 properly, I'd also argue against backporting #1212 on its own. Without the container side, the flag would stop crashing and start being silently ignored, which is worse than the present failure for exactly the reason this issue was opened: it is advertised in `--help`, so people believe it.

There is a workaround that works today on the affected distros, and as far as I can tell it isn't written down anywhere. Per-logger levels given to the container apply to components loaded into it afterwards, matched on node name:

```
ros2 run rclcpp_components component_container --ros-args --log-level mynode:=debug
ros2 component load /ComponentManager <package> <plugin> -n mynode
```

I tested this on Iron with a C++ component that logs at INFO in its constructor. Started with `--log-level testattach:=warn`, the container loads the component and the INFO line is suppressed. Without it the line appears. Raising works as well as lowering since a node created after logging is configured picks up `:=debug`.

The limitations are worth stating: you need the node name in advance and have to pass `-n` to match it, and it has to be set when the container starts rather than at load time. So it is not a replacement for `--log-level` on `load`, but it does mean nobody has to wait for the backport to get a component's debug output.

If a small change is wanted for humble, jazzy and kilted that doesn't touch rclcpp, I think the honest one is to stop advertising an option that cannot work there i.e., a `choices=` restriction, or an error pointing at the above. Either beats the current behaviour, which on Iron isn't even the traceback from the original report but an abort, because the generated conversion code asserts rather than raises. An abort with no Python frames is considerably harder to work back from.

Happy to send that CLI-side change if you think it's worth having.