ros2/ros2cli #700: ros2 topic pub auto header

CourchesneA, 2022-03-09:
## Feature request

#### Feature description
ROS1 allowed automatic filling of header and timestamp using keywords 'now' and 'auto'. It would be great to have that functionality in ROS2

#### Implementation considerations
ex.
```
ros2 topic pub my_topic geometry_msgs/TwistStamped '{header: auto}'
```

#### Expected behavior
```
geometry_msgs.msg.TwistStamped(header=std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=1646834300, nanosec=280516750), frame_id=''), twist=geometry_msgs.msg.Twist(linear=geometry_msgs.msg.Vector3(x=0.0, y=0.0, z=0.0), angular=geometry_msgs.msg.Vector3(x=0.0, y=0.0, z=0.0)))
```
#### Actual behavior
`Failed to populate field: Value 'auto' is expected to be a dictionary but is a str`

fujitatomoya, 2022-03-11:
ROS 1 wiki: http://wiki.ros.org/ROS/YAMLCommandLine#Headers.2Ftimestamps, this sounds reasonable to me. I would like to hear from others.

closed 2022-09-07 by ivanpauno

clalancette, 2022-09-07:
@ivanpauno @esteve Could you add a note to the [Iron Irwini release page](https://github.com/ros2/ros2_documentation/blob/rolling/source/Releases/Release-Iron-Irwini.rst)?  That way we'll have it queued up for release time.

esteve, 2022-09-07:
@clalancette I just submitted https://github.com/ros2/ros2_documentation/pull/2998

clalancette, 2022-09-07:
> @clalancette I just submitted [ros2/ros2_documentation#2998](https://github.com/ros2/ros2_documentation/pull/2998)

Thank you!

PR #749 (2022-09-07): Expand auto to the current time when passed to a Header field
This PR enables expansion of `auto` to a generated `Header` whose `stamp` field is set to the current time. This also expands `now` to the current time if the field is of type `builtin_interfaces.msg.Time`

These changes match the same behavior that `rostopic` has:

http://wiki.ros.org/ROS/YAMLCommandLine#Headers.2Ftimestamps

Closes #700