ros2/ros2cli #705: :man_farmer: Test regressions on various `ros2cli` utilities on the rolling jobs

Blast545, 2022-04-05:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu jammy
- Installation type:
  - Source, buildfarm
- Version or commit hash:
  - main
- DDS implementation:
  - Connext
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
Run a rolling buildfarm job.

#### Expected behavior
All test passes.

#### Actual behavior
`ros2cli` related errors appear, see: 
https://build.ros2.org/view/Rci/job/Rci__nightly-release_ubuntu_jammy_amd64/31/#showFailuresLink

Those have been consistently failing since the rolling jammy are building successfully.

ivanpauno, 2022-04-06:
Fixed by https://github.com/ros2/rmw_connextdds/pull/80.
`build.ros2.org` tests should start passing after we do a new `rmw_connextdds` release.

closed 2022-04-06 by ivanpauno

clalancette, 2022-04-06:
> `build.ros2.org` tests should start passing after we do a new `rmw_connextdds` release.

Actually, that shouldn't be necessary.  As far as I understand, that Rci job checks out the sources and builds them, so it should get the updates overnight.  I guess we'll see :).

ivanpauno, 2022-04-06:
> Actually, that shouldn't be necessary. As far as I understand, that Rci job checks out the sources and builds them, so it should get the updates overnight. I guess we'll see :).

It checks the sources of `ros2cli`, but it uses the last released version of `rmw_connextdds`, so it should still be failing.
The [Rci__nightly-connext](https://build.ros2.org/view/Rci/job/Rci__nightly-connext_ubuntu_jammy_amd64/37/#showFailuresLink) job also has similar failures, and those ones are using the released versions of the packages (I think).

ivanpauno, 2022-04-06:
> Actually, that shouldn't be necessary. As far as I understand, that Rci job checks out the sources and builds them, so it should get the updates overnight. I guess we'll see :).

Ah, I thought the ci jobs in `build.ros2.org` used the released version of the packages.