ros2/ros2cli #712: Tab completion doesn't work with cmd.exe or PowerShell

sloretz, 2022-05-11:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - binaries
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

In either cmd.exe or Powershell, activate the ROS 2 workspace

```
# Cmd.exe
call C:\dev\ros2_humble\setup.bat
# Powershell
. C:\dev\ros2_humble\setup.ps1
```

Type
```
ros2 lau<tab>
```

#### Expected behavior

The command would complete to `ros2 launch`

#### Actual behavior

Nothing happens

#### Additional information

sloretz, 2022-05-11:
Table completion on windows doesn't work in `Galactic` either, so maybe it's never worked on Windows.

clalancette, 2022-05-11:
> Table completion on windows doesn't work in `Galactic` either, so maybe it's never worked on Windows.

Thanks for checking on that.  Still a bug (tab completion should work), but not a Humble regression.