ros2/ros2cli #713: ros2 node list hangs when a node is on another machine on Windows and ROS_DOMAIN_ID is used

sloretz, 2022-05-11:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  -  binaries
- Version or commit hash:
  -  https://ci.ros2.org/view/packaging/job/packaging_windows/2478/
- DDS implementation:
  -  default
- Client library (if applicable):
  -  N/A

#### Steps to reproduce issue

On a linux laptop (not sure OS is important), set a ROS_DOMAIN_ID and start turtlesim_node

```
export ROS_DOMAIN_ID=42
ros2 run turtlesim turtlesim_node
```

On a Windows machine using the Binary archive, try to list the node

```
set ROS_DOMAIN_ID=42
ros2 node list
```

#### Expected behavior

The windows machine would list the turtlesim node.

#### Actual behavior

The `ros2 node list` command hangs forever, and does not respond to `CTRL + C`.

#### Additional information

I'm using a Windows VM on Virtual Box on Ubuntu Focal with bridged networking (machine A). The other machine (machine B) is an Ubuntu Bionic running ROS 2 nightly in a Jammy Apptainer container. I'm able to ping the other machine from the windows VM.

Running a Jammy Apptainer container on A, `ros2 node list` works as expected. It's only the Windows machine that can't list the node.

I could try a Humble workspace on machine B, but it's probably still similar enough.

`ros2 node list` returns immediately on the Windows VM when no `ROS_DOMAIN_ID` is set.

sloretz, 2022-05-11:
`ros2 topic list`, `ros2 service list`, and `ros2 param list` all hang in this scenario as well.

danzimmerman, 2022-09-16:
> It's only the Windows machine that can't list the node.

Can confirm I'm having this problem at least on some machines, but not all.

Is there any fix/workaround/suggestions for further debugging that would be useful?

All of my machines are using [Robostack](https://robostack.github.io/) conda environments.

The broken one is using `ros2cli 0.18.3`:

```
(humble) c:\Code\dbgdump>mamba list | findstr ros2cli
ros-humble-ros2cli        0.18.3           py39h4c0ac80_0    robostack-humble
ros-humble-ros2cli-common-extensions 0.1.1            py39he8739fe_0    robostack-humble
```
[mamba_exact_broken_win11.yaml.txt](https://github.com/ros2/ros2cli/files/9580801/mamba_exact_broken_win11.yaml.txt)

The full package plan is attached.

danzimmerman, 2022-09-16:
The working one is also `0.18.3`

The Linux machine with all the running nodes listed in the details below is an Ubuntu Focal machine also running Humble via Robostack.

```
(humble) c:\code>mamba list | findstr ros2cli
ros-humble-ros2cli        0.18.3           py39h4c0ac80_0    robostack-humble
ros-humble-ros2cli-common-extensions 0.1.1            py39he8739fe_0    robostack-humble
```

Package list attached.

[mamba_exact_working_win11.yaml.txt](https://github.com/ros2/ros2cli/files/9580817/mamba_exact_working_win11.yaml.txt)

**Results `ros2 topic list`**

<details>

```
(humble) c:\code>ros2 topic list
/attached_collision_object
/collision_object
/display_contacts
/display_cost_sources
/display_planned_path
/dynamic_joint_states
/force_torque_sensor_broadcaster/transition_event
/force_torque_sensor_broadcaster/wrench
/forward_position_controller/commands
/forward_position_controller/transition_event
/io_and_status_controller/io_states
/io_and_status_controller/robot_mode
/io_and_status_controller/robot_program_running
/io_and_status_controller/safety_mode
/io_and_status_controller/tool_data
/io_and_status_controller/transition_event
/joint_state_broadcaster/transition_event
/joint_states
/monitored_planning_scene
/motion_plan_request
/parameter_events
/planning_scene
/planning_scene_world
/robot_description
/robot_description_semantic
/rosout
/scaled_joint_trajectory_controller/joint_trajectory
/scaled_joint_trajectory_controller/state
/scaled_joint_trajectory_controller/transition_event
/servo_node/collision_velocity_scale
/servo_node/condition
/servo_node/delta_joint_cmds
/servo_node/delta_twist_cmds
/servo_node/publish_planning_scene
/servo_node/status
/servo_node/worst_case_stop_time
/speed_scaling_state_broadcaster/speed_scaling
/speed_scaling_state_broadcaster/transition_event
/tf
/tf_static
/trajectory_execution_event

(humble) c:\code>set | findstr ROS
ROS_DISTRO=humble
ROS_DOMAIN_ID=17
ROS_ETC_DIR=C:\Users\<username>\mambaforge\envs\humble\Library\etc\ros
ROS_LOCALHOST_ONLY=0
ROS_OS_OVERRIDE=conda:win64
ROS_PYTHON_VERSION=3
ROS_VERSION=2
```
</details>

**Machine Details**

<details>

```
Edition	Windows 11 Home
Version	21H2
Installed on	‎11/‎13/‎2021
OS build	22000.856
Experience	Windows Feature Experience Pack 1000.22000.856.0
```
</details>

cmastalli, 2023-06-29:
> Can confirm I'm having this problem at least on some machines, but not all.

I confirm this issue in my MacBook Pro M2 machine. I am also using [Robostack](https://robostack.github.io/GettingStarted.html). Are there any updates regarding this issue?

130s, 2023-06-30:
I know this ticket is for external-host scenario though, on the same single host scenario I just saw a similar issue. Easily reproducible with a Docker image (see "Steps to reproduce" section in https://github.com/ros-planning/moveit2_tutorials/issues/713).

MannavaVivek, 2024-06-14:
I am still facing this issue. Running ROS2 Humble on M3 Macbook Pro, using robostack. Tried with and without ROS_DOMAIN_ID. Thing is normally publishing and subscribing works, but when you give any of the list or echo commands, or even ros2 daemon commands, it just hangs until timeout. Any updates on this issue?

fujitatomoya, 2024-06-14:
could be related to https://github.com/ros2/ros2cli/issues/903.

if this is hitting https://github.com/ros2/ros2cli/pull/888, it should be fixed with next iron and humble patch release.

but i am not 100% sure about that. if you are willing to try https://github.com/ros2/ros2cli/pull/888 to see if that solves the problem. that would be really appreciated.

MannavaVivek, 2024-06-14:
> but i am not 100% sure about that. if you are willing to try #888 to see if that solves the problem. that would be really appreciated.

If by trying, you meant replacing my daemon.py with the one with the fixes here, I tried that but it doesn't seem to be working.

fujitatomoya, 2024-06-14:
> If by trying, you meant replacing my daemon.py with the one with the fixes here, I tried that but it doesn't seem to be working.

for fixing code, you are right.

just in case, can you restart the ros2 daemon.

```bash
### restart ros2 daemon to apply the change
ros2 daemon restart
```

and then check if the problem still occurs?

MannavaVivek, 2024-06-14:
This is one of the issues. I can't stop the daemon because the daemon stop command freezes as well. But a good old system restart did the trick