ros2/ros2cli #714: Autocomplete doesn't work with Ubuntu 20.04

TemugeB, 2022-05-16:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu20.04
- Installation type:
  apt install ros-foxy-desktop
- Version or commit hash:
 (0.9.11-1focal.20220209.145952)
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
ros2 run [TAB]
```

[image]

#### Expected behavior
Pressing [TAB] should show available packages. Unless I misunderstood and ros2 does not have tab completion.

#### Actual behavior
It only shows my current folder contents.

#### Additional information

Please check the screen shot for which commands I used. It can find and run packages fine if I type them out manually. But pressing [TAB] only shows current folder contents and does not show autocompleted packages.  

I also have noetic installed on the same machine. Noetic is not being sourced.

TemugeB, 2022-05-16:
ros2 doctor gives the follow error:
```
/opt/ros/foxy/lib/python3.8/site-packages/ros2doctor/api/__init__.py: 137: UserWarning: Fail to call TopicReport class functions.
````

The full ros2 doctor report is below.

```
ros2 wtf -r
/opt/ros/foxy/lib/python3.8/site-packages/ros2doctor/api/__init__.py: 137: UserWarning: Fail to call TopicReport class functions.

   NETWORK CONFIGURATION
ether        : MASKED
device       : enp3s0
flags        : 4099<UP,BROADCAST,MULTICAST> 
mtu          : 1500
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<UP,LOOPBACK,RUNNING> 
mtu          : 65536
inet         : 192.168.102.100
inet4        : ['192.168.102.100']
ether        : MASKED
inet6        : MASKED
netmask      : 255.255.255.0
device       : wlp5s0
flags        : 4163<UP,BROADCAST,RUNNING,MULTICAST> 
mtu          : 1500
broadcast    : 192.168.102.255

   PACKAGE VERSIONS
rviz_rendering                            : required=8.2.6, local=8.2.6
tf2_eigen                                 : required=0.13.13, local=0.13.13
examples_rclpy_minimal_action_server      : required=0.9.4, local=0.9.4
message_filters                           : required=3.2.5, local=3.2.5
rclcpp_components                         : required=2.4.1, local=2.4.1
pendulum_control                          : required=0.9.3, local=0.9.3
rviz_assimp_vendor                        : required=8.2.6, local=8.2.6
ament_package                             : required=0.9.5, local=0.9.5
osrf_pycommon                             : required=0.1.11, local=0.1.11
examples_rclcpp_minimal_subscriber        : required=0.9.4, local=0.9.4
ament_cmake_flake8                        : required=0.9.6, local=0.9.6
joy                                       : required=3.0.1, local=3.0.1
examples_rclpy_minimal_client             : required=0.9.4, local=0.9.4
teleop_twist_keyboard                     : required=2.3.2, local=2.3.2
rqt_graph                                 : required=1.1.3, local=1.1.3
logging_demo                              : required=0.9.3, local=0.9.3
ros_environment                           : required=2.5.0, local=2.5.0
tf2_sensor_msgs                           : required=0.13.13, local=0.13.13
rviz_common                               : required=8.2.6, local=8.2.6
rosbag2_transport                         : required=0.3.9, local=0.3.9
rmw_fastrtps_cpp                          : required=1.3.0, local=1.3.0
statistics_msgs                           : required=1.0.0, local=1.0.0
fastrtps_cmake_module                     : required=1.0.4, local=1.0.4
quality_of_service_demo_cpp               : required=0.9.3, local=0.9.3
domain_coordinator                        : required=0.9.2, local=0.9.2
rosbag2_storage                           : required=0.3.9, local=0.3.9
ament_cppcheck                            : required=0.9.6, local=0.9.6
pendulum_msgs                             : required=0.9.3, local=0.9.3
composition                               : required=0.9.3, local=0.9.3
tlsf                                      : required=0.5.0, local=0.5.0
rpyutils                                  : required=0.2.0, local=0.2.0
rosidl_cmake                              : required=1.2.1, local=1.2.1
tinyxml_vendor                            : required=0.8.2, local=0.8.2
ament_index_cpp                           : required=1.1.0, local=1.1.0
ament_cmake_ros                           : required=0.9.2, local=0.9.2
examples_rclcpp_minimal_timer             : required=0.9.4, local=0.9.4
[... 212 lines cut ...]
rqt_py_console                            : required=1.0.2, local=1.0.2
examples_rclcpp_minimal_service           : required=0.9.4, local=0.9.4
rqt_service_caller                        : required=1.0.5, local=1.0.5
ros2topic                                 : required=0.9.11, local=0.9.11
examples_rclpy_minimal_action_client      : required=0.9.4, local=0.9.4
geometry_msgs                             : required=2.0.5, local=2.0.5
rmw_implementation                        : required=1.0.3, local=1.0.3
intra_process_demo                        : required=0.9.3, local=0.9.3
rqt_gui_cpp                               : required=1.1.2, local=1.1.2

   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-5.13.0-41-generic-x86_64-with-glibc2.29
release          : 5.13.0-41-generic
processor        : x86_64

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : foxy
distribution type      : ros2
distribution status    : active
release platforms      : {'ubuntu': ['focal']}
```

clalancette, 2022-05-16:
Interesting that it isn't working for you.  I tried this out locally, and autocomplete seems to be working OK for me.

Can you run `dpkg -l | grep argcomplete` and paste the output here?

TemugeB, 2022-05-17:
Hi.

Thank you for replying to the issue. Your suggestion was right. This fixed the problem for me. Thank you.

```apt install python3-argcomplete```

closed 2022-05-17 by TemugeB