ros2/ros2cli #715: `pkg create --build-type ament_python` should warn about hyphens in pkg-name

krsche, 2022-05-21:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binaries
- Version or commit hash:
  - humble
- DDS implementation:
  - /
- Client library (if applicable):
  - ros2cli

#### Steps to reproduce issue
```
# this breaks
cd src
ros2 pkg create --build-type ament_python --license MIT --node-name hello-world  hello-world-python-1      # build breaks
ros2 pkg create --build-type ament_python --license MIT --node-name hello_world  hello-world-python-2     # build breaks
ros2 pkg create --build-type ament_python --license MIT --node-name hello-world  hello_world_python_3    # build breaks
ros2 pkg create --build-type ament_python --license MIT --node-name hello_world  hello_world_python_4   # build works
cd ..
colcon build --packages-select hello-world-python-1
colcon build --packages-select hello-world-python-2
colcon build --packages-select hello_world_python_3
colcon build --packages-select hello_world_python_4
```

#### Expected behavior
When running `ros2 pkg create` with `--build-type ament_python` warn about hyphens in the name / or abort pkg creation.

#### Actual behavior
No warning

#### Additional information
Not really a bug of `ros2cli` but also not really a feature request. I'd expect a warning, thats why I put it as a bug report.  
I assume the check can easily be added [here](https://github.com/ros2/ros2cli/blob/f4e5952f430e502060594d68f3a050b785fd4249/ros2pkg/ros2pkg/verb/create.py#L143).

If there is agreement on that, I can open a PR with it. Please let me know :)

krsche, 2022-05-22:
https://answers.ros.org/question/368144/colcon-build-fails-on-newly-created-package/?answer=396325#post-id-396325

aprotyas, 2022-05-24:
I agree that a warning about the hyphen would be informative here!

clalancette, 2022-05-24:
For what it is worth, while using underscore is our best practice, I don't think hyphens should fail.  I would prefer we figure out why hyphens are having problems.

clalancette, 2022-05-24:
> I think what's happening is that when you invoke `ros2 pkg create --build-type ament_python --license MIT --node-name hello-world hello-world-python-1`, a module named `hello-world` is created. I don't know where exactly it fails, but from what I understand modules with hyphens in the name can't be loaded/imported.

Oh, I see!  Thanks for the explanation.  OK, in that case I do agree that blocking them completely at `ros2 pkg create` time makes sense (for `ament_python` packages, at least).

(as a side note, I do believe that we can have hyphenated C++ packages, so for now I wouldn't change that semantic)

808brick, 2022-06-27:
For what it's worth, it seems the issue you may be facing is not from the package name, but the setupcfg file that get's autogenerated. 

It seems when the command is run, the setup.cfg file is created like so: 
```
[develop]
script-dir=$base/lib/hello-world-python
[install]
install-scripts=$base/lib/hello-world-python
```

When it should be :
```
[develop]
script_dir=$base/lib/hello-world-python
[install]
install_scripts=$base/lib/hello-world-python
```

Where the difference is a underscore vs the hyphen. (Should be `script_dir` vs the autogenerated `script-dir`)

I hope this is helpful and related to resolving your issue.