ros2/ros2cli #731: `ros2 topic echo` missing required positional argument `info` for

torrhen, 2022-07-02:
## Bug report

**Required Info:**
- Operating System:
Ubuntu 20.04.4 LTS
- Installation type:
Source
- Version or commit hash:
f46fa80c5bffe95b45fbba2e5cda724f3e4e8cce
- DDS implementation:
Cyclone DDS
- Client library (if applicable):
rclpy

#### Steps to reproduce issue
Caused when attempting to print the messages sent/received over a topic. The example below highlights this with a `turtlesim` example, but the same error is reproducible for `talker` and `listener` nodes over the `/chatter` topic.

`ros2 topic echo /turtle1/cmd_vel`

#### Expected behavior
```
linear:
  x: 2.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0
---
```
#### Actual behavior
```
Traceback (most recent call last):
  File "XXXX/install/ros2cli/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli', 'console_scripts', 'ros2')()
  File "XXXX/build/ros2cli/ros2cli/cli.py", line 89, in main
    rc = extension.main(parser=parser, args=args)
  File "XXXX/build/ros2topic/ros2topic/command/topic.py", line 41, in main
    return extension.main(args=args)
  File "XXXX/build/ros2topic/ros2topic/verb/echo.py", line 243, in main
    self.subscribe_and_spin(
  File "XXXX/build/ros2topic/ros2topic/verb/echo.py", line 286, in subscribe_and_spin
    rclpy.spin(node)
  File "XXXX/install/rclpy/lib/python3.8/site-packages/rclpy/__init__.py", line 222, in spin
    executor.spin_once()
  File "XXXX/install/rclpy/lib/python3.8/site-packages/rclpy/executors.py", line 712, in spin_once
    raise handler.exception()
  File "XXXX/install/rclpy/lib/python3.8/site-packages/rclpy/task.py", line 239, in __call__
    self._handler.send(None)
  File "XXXX/install/rclpy/lib/python3.8/site-packages/rclpy/executors.py", line 418, in handler
    await call_coroutine(entity, arg)
  File "XXXX/install/rclpy/lib/python3.8/site-packages/rclpy/executors.py", line 343, in _execute_subscription
    await await_or_execute(sub.callback, msg)
  File "XXXX/rclpy/lib/python3.8/site-packages/rclpy/executors.py", line 107, in await_or_execute
    return callback(*args)
TypeError: _subscriber_callback() missing 1 required positional argument: 'info'
```
#### Additional information
The same error is thrown when including the information of a message over the topic, e.g.
`ros2 topic echo /turtle1/cmd_vel --include-message-info --csv`

A temporary fix can be found by providing a default argument to the `_subscriber_callback` method below on line 288 within ros2topic/verb/echo.py
```
    def _subscriber_callback(self, msg, info='Default'):
        submsg = msg
        if self.field is not None:
            for field in self.field:
                try:
                    submsg = getattr(submsg, field)
                except AttributeError as ex:
                    raise RuntimeError(f"Invalid field '{'.'.join(self.field)}': {ex}")

```
This gives the expected behaviour for the ordinary `ros2 topic echo <topic-name>` example above as well for `ros2 topic echo --include-message-info <topic-name>` with or without `--raw` (which prints the 'Default' string before each message in this case). 

However, this does not help with `--include-message-info --csv` as it raises an entirely new error.

fujitatomoya, 2022-07-06:
I cannot reproduce this issue on Ubuntu 20.04 or 22.04 with https://github.com/ros2/ros2/commit/28a83a843324918293ccd0d106117cf2fa4e33a6 (cyclonedds and fastdds just works okay.)

torrhen, 2022-07-06:
New install and build resolved the issue in the end. Thanks for checking regardless.

closed 2022-07-06 by torrhen