ros2/ros2cli #738: ros2 topic delay for tf

timonegk, 2022-07-15:
## Feature request

#### Feature description
It would be nice to be able to run ros2 topic delay for the `/tf` topic. It has the message type `tf2_msgs/msg/TFMessage` which is a list of `TransformStamped`, each containing a header.

#### Implementation considerations
Either messages of type `tf2_msgs/msg/TFMessage` could always be treated in this way or a command line option could be added that makes it possible to give a subfield that should be checked for headers.

fujitatomoya, 2022-07-20:
I think `ros2 topic delay` is meant to measure and display the delay of messages via topic, not contents in the message. but i would like to hear from others.

timonegk, 2022-07-21:
If I understand it correctly, `ros2 topic delay` reads the contents of the message's header and determines the difference to the current clock. So it does already use contents of the message. My proposal would just be to use different fields for the `/tf` topic.

clalancette, 2022-08-04:
I don't think we should special-case `TFMessage` here, but I would be for a new command-line option to allow the field to be chosen.

That said, I don't think we'll get to this anytime soon, so please consider making a pull request to implement this functionality.