ros2/ros2cli #742: Message template autocompletion broken

Flova, 2022-07-29:
When publishing a topic via `ros2 topic pub` or calling a service via `ros2 service call` the tab completion for the fields of the message is broken. The completion for the topic and message type works fine. The issue seems to be different between zsh and bash.

In zsh:
The field of the message are all handled are possible cli parameters that can be autocompleted pressing tab iterates through the fields. Each of the fields has its escaped indentation in the beginning and is individually selectable. Creating a valid message is impossible even for small msgs.
Example: 
```zsh
🐋 /home/florian/Projekt/bitbots/colcon_ws  ros2 topic pub /camera_info sensor_msgs/msg/CameraInfo -\ 0.0\
-\ 0.0\                        \ \ do_rectify:\ false\        --keep-alive                   p:\                            --qos-reliability              \ \ stamp:\                    width:\ 0\                   
-1                             \ \ frame_id:\ \'\'\           -n                             --print                        -r                             -t                             \ \ x_offset:\ 0\            
binning_x:\ 0\                 header:\                       \ \ \ \ nanosec:\ 0\           --qos-depth                    r:\                            --times                        \ \ y_offset:\ 0\            
binning_y:\ 0\                 \ \ height:\ 0\                --node-name                    --qos-durability               --rate                         -w                                                          
d:\ []\                        height:\ 0\                    --once                         --qos-history                  roi:\                          --wait-matching-subscriptions                               
distortion_model:\ \'\'\       k:\                            -p                             --qos-profile                  \ \ \ \ sec:\ 0\               \ \ width:\ 0\                                              
```

In bash:
Bash looks overall better as the message template is printed as one piece, but the completed text is not editable. Instead, the user is thrown into a new terminal line containing the command before the tab completion. 

Example:
```bash
[florian@flovat colcon_ws]$ ros2 topic pub /camera_info sensor_msgs/msg/CameraInfo 
-1
header:\^J\ \ stamp:\^J\ \ \ \ sec:\ 0\^J\ \ \ \ nanosec:\ 0\^J\ \ frame_id:\ \'\'\^Jheight:\ 0\^Jwidth:\ 0\^Jdistortion_model:\ \'\'\^Jd:\ []\^Jk:\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^Jr:\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^Jp:\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^J-\ 0.0\^Jbinning_x:\ 0\^Jbinning_y:\ 0\^Jroi:\^J\ \ x_offset:\ 0\^J\ \ y_offset:\ 0\^J\ \ height:\ 0\^J\ \ width:\ 0\^J\ \ do_rectify:\ false\
--keep-alive
-n
--node-name
--once
-p
--print
--qos-depth
--qos-durability
--qos-history
--qos-profile
--qos-reliability
-r
--rate
-t
--times
-w
--wait-matching-subscriptions
[florian@flovat colcon_ws]$ ros2 topic pub /camera_info sensor_msgs/msg/CameraInfo
```

I observed this behavior on all ros2 machines I have used so far, making the cli unusable for publishing topics.

## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04 LTS
- Installation type:
  - binaries 
- Version or commit hash:
  - 0.19.0-1jammy.20220711.162709
- DDS implementation:
  - Cyclone & FastDDS

#### Steps to reproduce issue
- source your terminal (zsh or bash)
- type `ros2 topic pub /camera_info sensor_msgs/msg/CameraInfo` and press `TAB`

#### Expected behavior
A nicely formatted and editable string containing the message fields with their defaults should appear. 

#### Actual behavior
Either the message is split and autocompleted in pieces (zsh) or not editable (bash)

#### Implementation considerations
I am not familiar with terminal formatting.

Flova, 2022-07-29:
ROS 1 completion resulted in:

```zsh
rostopic pub /cmd_vel geometry_msgs/Twist "linear:                 
  x: 0.05
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0"

```

clalancette, 2022-07-29:
So this isn't ideal, but at least in bash if you start with a `"`, and then the first letter of the message, then autocompletion works as expected:

```
$ . /opt/ros/rolling/setup.bash 
$ ros2 topic pub /camera_info sensor_msgs/msg/CameraInfo "<TAB>
-1
header:^J  stamp:^J    sec: 0^J    nanosec: 0^J  frame_id: ''^Jheight: 0^Jwidth: 0^Jdistortion_model: ''^Jd: []^Jk:^J- 0.0^J- 0.0^J- 0.0^J- 0.0^J- 0.0^J- 0.0^J- 0.0^J- 0.0^J- 0.0^Jr:^J- 0.0^J- 0.0^J- 0.0^J- 0.0^J- 0.0^J- 0.0^J- 0.0^J- 0.0^J- 0.0^Jp:^J- 0.0^J- 0.0^J- 0.0^J- 0.0^J- 0.0^J- 0.0^J- 0.0^J- 0.0^J- 0.0^J- 0.0^J- 0.0^J- 0.0^Jbinning_x: 0^Jbinning_y: 0^Jroi:^J  x_offset: 0^J  y_offset: 0^J  height: 0^J  width: 0^J  do_rectify: false
--keep-alive
-n
--node-name
--once
-p
--print
--qos-depth
--qos-durability
--qos-history
--qos-profile
--qos-reliability
-r
--rate
-t
--times
-w
--wait-matching-subscriptions
$ ros2 topic pub /camera_info sensor_msgs/msg/CameraInfo "h<TAB>
$ ros2 topic pub /camera_info sensor_msgs/msg/CameraInfo "header:
  stamp:
    sec: 0
    nanosec: 0
  frame_id: ''
height: 0
width: 0
distortion_model: ''
d: []
k:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
r:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
p:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
binning_x: 0
binning_y: 0
roi:
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: false"
```

It would be fantastic to figure out how to do that by default, so we don't need those beginning two letters.

Flova, 2022-07-29:
We could also use something without newlines (maybe json), as this most likely breaks the autocomplete. It seems like most autocompletion systems don't play well with long multi line strings. Maybe the autocompletion could be dropped for this argument, and we add an "interactive mode" where the user is able to edit the msg after sending the command.

Flova, 2022-07-29:
One could just leave the values field empy and a https://github.com/prompt-toolkit/python-prompt-toolkit multiline prompt is shown.

Flova, 2022-07-29:
Editing the msg like this would be fancy, try the snipped:

```python
from prompt_toolkit import prompt
from pygments.lexers.data import YamlLexer
from prompt_toolkit.lexers import PygmentsLexer

prompt("> ", multiline=True, default=
 """header:
   stamp:
     sec: 0
     nanosec: 0
   frame_id: ''
 height: 0
 width: 0
 distortion_model: ''
 d: []
 k:
 - 0.0
 - 0.0
 - 0.0
 - 0.0
 - 0.0
 - 0.0
 - 0.0
 - 0.0
 - 0.0
 r:
 - 0.0
 - 0.0
 - 0.0
 - 0.0
 - 0.0
 - 0.0
 - 0.0
 - 0.0
 - 0.0
 p:
 - 0.0
 - 0.0
 - 0.0
 - 0.0
 - 0.0
 - 0.0
 - 0.0
 - 0.0
 - 0.0
 - 0.0
 - 0.0
 - 0.0
 binning_x: 0
 binning_y: 0
 roi:
   x_offset: 0
   y_offset: 0
   height: 0
   width: 0
   do_rectify: false""", lexer=PygmentsLexer(YamlLexer))
```

Both deps are commonly used, in the Ubuntu repos and cross platform, but the syntax highlighting is not that nessesary if you want to keep the deps low.

clalancette, 2022-07-29:
True.  This even seems to work in Windows.

I'd be for adding a mode like this; I think it would be easier to use.  I don't necessarily think we should make it the default (it is not useful for scripting), but for manual interaction it would be pretty cool.

leander-dsouza, 2025-03-29:
I would like to take this issue, please :)

christophebedard, 2025-03-29:
Great! I've assigned the issue to you, let us know if you have any questions.

leander-dsouza, 2025-03-31:
Hey @christophebedard , I have taken a stab at this in #995 :)

BruceChanJianLe, 2026-01-07:
> that looks great! Is there a way this fix could be made here in `ros2cli` to aovid relying on a separate ansible-based patch?

Thanks for the suggestion, I surely can push a fix for this, but from my opinion, this is somewhat of a hack, and I'm not sure if everyone is agreeable with the approach. Hence, I did not make a pull request.

Basically, since the colon ":" character is causing confusion for zsh, I swap it from the equal "=" character. And before publishing it with pub, it will swap back to colon (back to yaml format) and then publishes it out.

Let me know what your thoughts are, I will be happy to create a pull request :D