ros2/ros2cli #748: `get_action_names_and_types` is not implemented by the rclpy node class

alsora, 2022-08-17:
## Bug report

The `get_action_names_and_types` API defined here https://github.com/ros2/ros2cli/blob/rolling/ros2action/ros2action/api/__init__.py calls an homonym method on the `Node` rclpy class.
It looks like this function is not implemented.

This problem applies to both Galactic and Humble.

Is there a reason for that?
Which of the two libraries (`ros2cli` or `rclpy`) should be updated?

alsora, 2022-08-18:
Interesting, I didn't see that cpp/python binding, however the rclpy node python class does not expose that function.
We have the similar functions for topics and services, but not the one for actions.

https://github.com/ros2/rclpy/blob/rolling/rclpy/rclpy/node.py#L1878

fujitatomoya, 2022-08-18:
> rclpy node python class does not expose that function.

(`rclpy.action` does) but agree, this seems to be inconsistent for me.

clalancette, 2026-11-14:
I'm pretty sure that this is now implemented via ros2/rclpy#1574 and #1188 here, so closing this out.

closed 2026-11-14 by clalancette