ros2/ros2cli #752: [ros2 topic hz/bw/pub] support use_sim_time

fujitatomoya, 2022-08-31:
## Feature request

#### Feature description

- `hz`, calculate average publish rate based on simulation time.
- `bw`, calculate bandwidth based on simulation time.
- `pub`, use simulation time for `rate` option.

#### Implementation considerations

follow up from https://github.com/ros2/ros2cli/issues/515

llapx, 2022-09-09:
Hi all, 

the fix is completed, thanks for review.

fujitatomoya, 2022-09-09:
@clalancette @ivanpauno @wjwwood  what do you think about these extension? these are come from our actual use cases, can you share your thoughts?

ivanpauno, 2022-09-09:
+1
In ros 1 `use_sim_time` was set in the parameter server, so `rostopic` automatically was able to detect if sim time needed to be used.
Now we will need an explicit way to ask for sim time to be used in `ros2 topic`.

We could also by default check if there's someone publishing to `/clock`, and set `use_sim_time` based on that by default, but I'm not sure that's a good idea or not.

I would start by adding the option of explicitly setting simulation time, and later on we can see if the default behavior is changed or not.

fujitatomoya, 2022-09-21:
https://github.com/ros2/ros2cli/pull/754 is merged, close this issue.

closed 2022-09-21 by fujitatomoya