ros2/ros2cli #755: ros2 cli wrong help message for argument

omertal88, 2022-09-06:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
ros2 component load -h
```

#### Expected behavior
`container_node_name   Container node name to unload component from`
#### Actual behavior
`container_node_name   Container node name to load component to`
(please correct my grammer if needed).
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

clalancette, 2022-09-06:
Good call, see #756

closed 2022-09-06 by clalancette

fujitatomoya, 2022-09-06:
@clalancette do we want to backport this to humble/galactic/foxy?

clalancette, 2022-09-06:
> @clalancette do we want to backport this to humble/galactic/foxy?

Yeah, good call.

clalancette, 2022-09-06:
@Mergifyio backport humble galactic foxy

PR #756 (2022-09-06): Fix the component load help to mention load, not unload.

Fixes #755