ros2/ros2cli #766: why is there no option 'info' in ros2 service ?

leeminju531, 2022-09-22:
there in no option 'info' in ros2 service. although less than topics, I sometimes want to know service clients,server counts like topics.
could you have any plan to implement this? 
thank you

clalancette, 2022-09-22:
In short, we've never had a request for it before.  You can get some of the information you want by running `ros2 node info`, which lists the service clients and servers that are available, along with their types.

That said, this is a valid feature request.  We probably won't have time to work on it, but if you are interested in doing the work for it and making pull requests, we'd be happy to review.

leeminju531, 2022-09-22:
okay, it's my pleasure. I'll see it.

schornakj, 2022-12-15:
I previously submitted a PR to add this feature in #703. Certainly agreed that it's a very useful tool for debugging.

fujitatomoya, 2024-01-19:
https://github.com/ros2/ros2cli/pull/771 is completed, i will go ahead to close this issue.

closed 2024-12-27 by clalancette