ros2/ros2cli #768: ros2 topic echo array for covariance wrong format

hansryant, 2022-10-03:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- DDS implementation:
  - RTI Connext
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

```
ros2 topic echo /imu/data_raw
```

#### Expected behavior
Running in Ubuntu 18.04, ROS2 Eloquent, when I try to echo the topic:
```
header:
  stamp:
    sec: 1664780446
    nanosec: 971418804
  frame_id: imu_link
orientation:
  x: 0.0
  y: 0.0
  z: 0.0
  w: 1.0
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity:
  x: -0.01597896654047543
  y: -0.01597896654047543
  z: 0.006391586616190171
angular_velocity_covariance: [0.0017453292519943296, 0.0, 0.0, 0.0, 0.0017453292519943296, 0.0, 0.0, 0.0, 0.0017453292519943296]
linear_acceleration:
  x: -7.900865478515625
  y: -2.2936451904296873
  z: 5.391742138671875
linear_acceleration_covariance: [0.07845239999999999, 0.0, 0.0, 0.0, 0.07845239999999999, 0.0, 0.0, 0.0, 0.07845239999999999]
```
#### Actual behavior
And then I have another machine that is running in Ubuntu 20.04, ROS2 Foxy, and the topic I got from echo is:
```
header:
  stamp:
    sec: 1664783611
    nanosec: 622761912
  frame_id: imu_link
orientation:
  x: 0.0
  y: 0.0
  z: 0.0
  w: 1.0
orientation_covariance:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
angular_velocity:
  x: -0.019174759848570515
  y: -0.009587379924285258
  z: 0.005326322180158476
angular_velocity_covariance:
- 0.0017453292519943296
- 0.0
- 0.0
- 0.0
- 0.0017453292519943296
- 0.0
- 0.0
- 0.0
- 0.0017453292519943296
linear_acceleration:
  x: -6.64630380859375
  y: -2.437297290039062
  z: 6.852205151367187
linear_acceleration_covariance:
- 0.07845239999999999
- 0.0
- 0.0
- 0.0
- 0.07845239999999999
- 0.0
- 0.0
- 0.0
- 0.07845239999999999
```
#### Additional information
I want to replicate the topic format from the expected behavior when I run the ROS2 Foxy on it, but it seems that I can't find the problem or how to fix the formatting problem, especially in the covariance part (which is an array).  With the format of newline for each covariance is hard to read compared with the array format like the expected behavior. Is there any suggested solution to debug or to fix the issue? Thank you

fujitatomoya, 2022-10-03:
related to https://github.com/ros2/ros2cli/issues/680

clalancette, 2022-10-03:
I changed the title since the current format isn't wrong, though it may be inconvenient in some circumstances.

@hansryant I know for at least Rolling (and probably Humble) we added in a `--flow-style` argument so you can get the old argument (what @fujitatomoya references above).  Can you try that out and see if it is more to your liking?

hansryant, 2022-10-05:
I can only use eloquent since my system is only running ubuntu 18.04, but anyway I tried to follow the #680 implementation on how to solve this and it works!

I will just put the steps on how I do it in case other people have this issue as well (for older ROS2 distro):
1. go to `/opt/ros/<distro>/lib/python3.8/site-package/rosidl_runtime_py`
2. edit `convert.py`
3. find the `yaml.dump` and add `default_flow_style=True` to it

Thanks for the help! @fujitatomoya @clalancette

closed 2022-10-05 by hansryant