ros2/ros2cli #772: Flaky test on ros2topic: `test_use_sim_time.TestROS2TopicUseSimTime.test_pub_times`

Crola1702, 2022-10-17:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04 (Jammy)
- Installation type:
  - Source
- Version or commit hash:
  - Rolling
- DDS implementation:
  - All

#### Steps to reproduce issue

1. Wait for Ros2 ci nightlies to run
2. See test failure (Reference build [here](https://ci.ros2.org/view/nightly/job/nightly_linux_debug/2456/))

#### Expected behavior

Pass on all builds

#### Actual behavior

Failing in some builds (percentages for the last 2 weeks):
* [nightly_linux_debug](https://ci.ros2.org/view/nightly/job/nightly_linux_debug/2456/): 13.3% of the times
* [nightly_linux_repeated](https://ci.ros2.org/view/nightly/job/nightly_linux_repeated/2802/): 6.6% of the times
* [nightly_linux-rhel_debug](https://ci.ros2.org/view/nightly/job/nightly_linux-rhel_debug/1291/): 33.3% of the times
* [nightly_linux-rhel_release](https://ci.ros2.org/view/nightly/job/nightly_linux-rhel_release/1290/#showFailuresLink): 13.3% of the times
* [nightly_linux-rhel_repeated](https://ci.ros2.org/view/nightly/job/nightly_linux-rhel_repeated/1284/): 13.3% of the times

Log output:
```
FAIL: test_use_sim_time.TestROS2TopicUseSimTime.test_pub_times
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jenkins-agent/workspace/nightly_linux_debug/ws/install/launch_testing/lib/python3.10/site-packages/launch_testing/markers.py", line 57, in _wrapper
    return func(self, *args, **kwargs)
  File "/home/jenkins-agent/workspace/nightly_linux_debug/ws/src/ros2/ros2cli/ros2topic/test/test_use_sim_time.py", line 124, in test_pub_times
    assert command.wait_for_shutdown(timeout=1)
AssertionError: assert False
 +  where False = <bound method ProcessProxy.wait_for_shutdown of <launch_testing.tools.process.ProcessProxy object at 0x7f5f3fac0d60>>(timeout=1)
 +    where <bound method ProcessProxy.wait_for_shutdown of <launch_testing.tools.process.ProcessProxy object at 0x7f5f3fac0d60>> = <launch_testing.tools.process.ProcessProxy object at 0x7f5f3fac0d60>.wait_for_shutdown
```

#### Additional information

First failing build (for linux): https://ci.ros2.org/view/nightly/job/nightly_linux_repeated/2802/
First failing build (for rhel): https://ci.ros2.org/view/nightly/job/nightly_linux-rhel_debug/1279/

Blast545, 2022-10-21:
Using scripting database tools, the first time it appeared in Rolling debug (30 days ago)

https://build.ros2.org/view/Rci/job/Rci__nightly-release_ubuntu_jammy_amd64/207/
https://ci.ros2.org/view/nightly/job/nightly_linux-rhel_debug/1268/

Crola1702, 2022-11-07:
It happened today in Linux Debug build: https://ci.ros2.org/view/nightly/job/nightly_linux_debug/

Crola1702, 2022-11-09:
It happened today in Linux Release build: https://ci.ros2.org/view/nightly/job/nightly_linux_release/2459/

Blast545, 2022-12-09:
This error was added with this PR: https://github.com/ros2/ros2cli/pull/754

fujitatomoya, 2022-12-15:
this flaky test has been addressed for humble and foxy, i will close this issue.

closed 2022-12-15 by fujitatomoya