ros2/ros2cli #778: No way to load a ros param file for all nodes

ckurtz22, 2022-11-10:
----
## Feature request

#### Feature description
ROS2 has no functionality for loading a parameter file that contains parameters for multiple nodes. This was easily doable in ROS1 and there is no reason it should not be possible in ROS2. For example, parameter file named "params.yaml" contains the following:

```/my_node1:
  ros__parameters:
    param1: 1.0
    param2: "test"
/my_node2:
  ros__parameters:
    param3: 73
```

and then running `ros2 param load params.yaml` would apply the relevant parameters to the nodes named `/my_node1` and `/my_node2`

#### Implementation considerations

fujitatomoya, 2022-11-10:
right, this cannot be done with current `ros2 param` command, since it requires `node_name` in the argument. I think this feature is reasonable.

CC: @ros2/team

RFRIEDM-Trimble, 2022-11-15:
This sounds like a fun one, I'd like to take the work if the team is happy with the modification. Seems straightforward enough.

reinzor, 2023-10-27:
Any updates on this; we could help with the implementation if this suggested feature is desired.

fujitatomoya, 2023-10-27:
No, i do not think anybody working on this. feature sounds good to me, but it could not be so straight-forward. currently it knows which node to set the parameters because user specifies so, but this requires to parse the parameter file to which node to call the parameter service (including wildcard), besides needs to create `ParameterClient` for each of them.