ros2/ros2cli #780: ros2interface can't show IDL messages

esteve, 2022-11-14:
## Bug report

**Required Info:**

- Operating System:
  - 

Ubuntu 22.041 LTS

- Installation type:
  - 

Binaries

- Version or commit hash:
  - 

f6fc61accff8b9648408053fb66b936839fe449d

- DDS implementation:
  - 

FastDDS

- Client library (if applicable):
  - 
N/A

#### Steps to reproduce issue

 - Have a package that contains messages in .IDL format in a workspace (e.g. https://gitlab.com/autowarefoundation/autoware.auto/autoware_auto_msgs/-/tree/master/)
 - Source the workspace
 - The messages show in `ros2 interface list`

```
$ ros2 interface list -m
Messages:
    action_msgs/msg/GoalInfo
    action_msgs/msg/GoalStatus
    action_msgs/msg/GoalStatusArray
    actionlib_msgs/msg/GoalID
    actionlib_msgs/msg/GoalStatus
    actionlib_msgs/msg/GoalStatusArray
    autoware_auto_control_msgs/msg/AckermannControlCommand
    autoware_auto_control_msgs/msg/AckermannLateralCommand
    autoware_auto_control_msgs/msg/HighLevelControlCommand
    autoware_auto_control_msgs/msg/LongitudinalCommand
    autoware_auto_debug_msgs/msg/BoolStamped
    autoware_auto_debug_msgs/msg/Float32MultiArrayStamped
    autoware_auto_debug_msgs/msg/Float32Stamped
    autoware_auto_debug_msgs/msg/Float64MultiArrayStamped
    autoware_auto_debug_msgs/msg/Float64Stamped
    autoware_auto_debug_msgs/msg/Int32MultiArrayStamped
    autoware_auto_debug_msgs/msg/Int32Stamped
    autoware_auto_debug_msgs/msg/Int64MultiArrayStamped
    autoware_auto_debug_msgs/msg/Int64Stamped
    autoware_auto_debug_msgs/msg/MultiArrayDimension
    autoware_auto_debug_msgs/msg/MultiArrayLayout
    autoware_auto_debug_msgs/msg/StringStamped
...
```
 - Show one of the IDL messages

```
$ ros2 interface show autoware_auto_control_msgs/msg/AckermannControlCommand
```

#### Expected behavior

`ros2 interface show` will show the message

#### Actual behavior

`ros2 interface show` throws an exception because it can't parse IDL messages:

```
$ ros2 interface show autoware_auto_control_msgs/msg/AckermannControlCommand
#include "autoware_auto_control_msgs/msg/AckermannLateralCommand.idl"
#include "autoware_auto_control_msgs/msg/LongitudinalCommand.idl"
#include "builtin_interfaces/msg/Time.idl"

Error processing 'module autoware_auto_control_msgs {' of 'autoware_auto_control_msgs/AckermannControlCommand': ''module' is an invalid message name. It should have the pattern '^[A-Z][A-Za-z0-9]*$''
Traceback (most recent call last):
  File "/home/esteve/Projects/tier4/ros2cli_ws/install/ros2cli/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.21.0', 'console_scripts', 'ros2')())
  File "/home/esteve/Projects/tier4/ros2cli_ws/install/ros2cli/lib/python3.10/site-packages/ros2cli/cli.py", line 89, in main
    rc = extension.main(parser=parser, args=args)
  File "/home/esteve/Projects/tier4/ros2cli_ws/install/ros2interface/lib/python3.10/site-packages/ros2interface/command/interface.py", line 35, in main
    return extension.main(args=args)
  File "/home/esteve/Projects/tier4/ros2cli_ws/install/ros2interface/lib/python3.10/site-packages/ros2interface/verb/show.py", line 201, in main
    _show_interface(
  File "/home/esteve/Projects/tier4/ros2cli_ws/install/ros2interface/lib/python3.10/site-packages/ros2interface/verb/show.py", line 148, in _show_interface
    for line in _get_interface_lines(interface_identifier):
  File "/home/esteve/Projects/tier4/ros2cli_ws/install/ros2interface/lib/python3.10/site-packages/ros2interface/verb/show.py", line 116, in _get_interface_lines
    yield InterfaceTextLine(
  File "/home/esteve/Projects/tier4/ros2cli_ws/install/ros2interface/lib/python3.10/site-packages/ros2interface/verb/show.py", line 43, in __init__
    msg_spec = parse_message_string(
  File "/home/esteve/Projects/tier4/ros2cli_ws/install/rosidl_adapter/local/lib/python3.10/dist-packages/rosidl_adapter/parser.py", line 520, in parse_message_string
    Type(type_string, context_package_name=pkg_name),
  File "/home/esteve/Projects/tier4/ros2cli_ws/install/rosidl_adapter/local/lib/python3.10/dist-packages/rosidl_adapter/parser.py", line 277, in __init__
    super(Type, self).__init__(
  File "/home/esteve/Projects/tier4/ros2cli_ws/install/rosidl_adapter/local/lib/python3.10/dist-packages/rosidl_adapter/parser.py", line 205, in __init__
    raise InvalidResourceName(
rosidl_adapter.parser.InvalidResourceName: 'module' is an invalid message name. It should have the pattern '^[A-Z][A-Za-z0-9]*$'
```

#### Additional information

fujitatomoya, 2022-11-14:
@esteve thanks for the issue, this is reproducible.

> Error processing 'module autoware_auto_control_msgs {' of 'autoware_auto_control_msgs/HighLevelControlCommand': ''module' is an invalid message name. It should have the pattern '^[A-Z][A-Za-z0-9]*$''

i think parser mistakes the argument with `module` according to the above log, and then raises `InvalidResourceName`.

fujitatomoya, 2022-11-14:
@sloretz do you have any thoughts for this issue?

esteve, 2022-11-15:
@fujitatomoya thanks. I have something in the works, but it requires changes in `rosidl_adapter` as well. Basically what happens is that `parse_message_string` (called from https://github.com/ros2/ros2cli/blob/rolling/ros2interface/ros2interface/verb/show.py#L43), does not understand IDL messages and there's no equivalent in `rosidl_adapter.parser`. There's a `parse_idl_string` in `rosidl_parser.parser` (see https://github.com/ros2/rosidl/blob/rolling/rosidl_parser/rosidl_parser/parser.py#L71), but what it does it parses an entire IDL message from a string, but `parse_message_string` parses a ROSmsg message line by line

Druthyn, 2023-08-21:
Is this still being looked at at all? I've just come across the issue myself.

clalancette, 2023-08-21:
> Is this still being looked at at all? I've just come across the issue myself.

We don't have anybody actively looking at it.  If you'd like to contribute support for this, we'd be happy to review it.

fujitatomoya, 2023-08-21:
CC: @iuhilnehc-ynos

iuhilnehc-ynos, 2023-08-22:
Currently, I think it's suggested to use ros2 types (".msg/.srv/.action") directly which could be adapted (parsered) to ".idl".

What the ros2 interface does right now is to show the original message with nested type and comments, which is similar to the original ".msg/.srv/.action" file.

I wondered what the expected output is after calling `ros2 interface show example_interfaces/srv/AddTwoIntsIDL` for the ".idl".

A new IDL file `AddTwoIntsIDL.idl` is used directly
```
module example_interfaces {
  module srv {
    struct AddTwoIntsIDL_Request {
      @verbatim(language = "comment", text =
        "Just a test comment for a")
      sequence < int64, 3 > a;

      int64 b;
    };
    struct AddTwoIntsIDL_Response {
      int64 sum;
    };
  };
};

```

What we expect the output of running `ros2 interface show example_interfaces/srv/AddTwoIntsIDL` as below,

```
# Just a test comment for a
int64[<=3] a
int64 b

---

int64 sum
```

Right?

clalancette, 2023-08-22:
> Right?

No, I don't think so.  The main reason is that it is possible to create `.idl` files that can't possibly be converted to `.msg` or `.srv` files.  That was the original intention behind allowing users to specify `.idl` files; so that they can use features that we don't have implemented in `.msg`.  We don't really support any of those features at present, but I can easily see that we might in the future.

For that reason, what I think `ros2 interface show` should do is display whatever format the original message is in.  If the original was a `.msg`, then we should display that.  If the original was a `.idl`, then we should display that.  Etc.

Nup-hsa, 2023-11-16:
I am trying the same and came with the identical error. Has this issue been solved? 
Thank you in advance!!!

esteve, 2023-11-16:
@Nup-hsa I haven't had time to work on this, AFAIK there's been no progress since I created this ticket.

johnchars, 2024-07-11:
I met the similar error
```shell
Error processing '// generated from rosidl_adapter/resource/msg.idl.em' of 'collect_msgs/Point2D': '//'
Traceback (most recent call last):
  File "/opt/ros/humble/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.18.10', 'console_scripts', 'ros2')())
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2cli/cli.py", line 91, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2interface/command/interface.py", line 35, in main
    return extension.main(args=args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2interface/verb/show.py", line 201, in main
    _show_interface(
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2interface/verb/show.py", line 148, in _show_interface
    for line in _get_interface_lines(interface_identifier):
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2interface/verb/show.py", line 116, in _get_interface_lines
    yield InterfaceTextLine(
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2interface/verb/show.py", line 43, in __init__
    msg_spec = parse_message_string(
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/parser.py", line 520, in parse_message_string
    Type(type_string, context_package_name=pkg_name),
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/parser.py", line 277, in __init__
    super(Type, self).__init__(
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/parser.py", line 190, in __init__
    raise InvalidResourceName(type_string)
rosidl_adapter.parser.InvalidResourceName: //
```

The reason is my msg directory has subdirectory like `msg/common/Point2D.msg`, i solve it by moving the Point2D.msg to msg. 
```shell
# change
msg/common/Point2D.msg
# to
msg/Point2D.msg
```
The magic thing is that before i create a srv/ and adding a Foo.srv, it's work.

awesomebytes, 2025-05-15:
I got the same error, but due to a different reason: I had softlinked some messages from a different folder, and the install was pointing to the wrong place (softlink to softlink points to the end-softlinj path).

Here I wrote my errors and workaround if someone runs into the same issue:
https://gist.github.com/awesomebytes/31ce0d00ceb308667b4b48e4a92fb15c