ros2/ros2cli #781: Feature Request: Provide CLI to retrieve IP address a node runs on

moooeeeep, 2022-12-05:
I recently was in the situation, that two nodes were publishing on the same topic (classroom environment with several turtlebots, two of which with colliding `ROS_DOMAIN_ID`), and it would have been helpful if there was a command to find the IP address of each node in the system (some kind of a network inventory). We ended up to go after the `tcpdump` approach [explained here](http://docs.ros.org/en/rolling/Tutorials/Advanced/Security/Examine-Traffic.html#display-unencrypted-discovery-packets) to identify the offending host. 

I thought having a ROS 2 native CLI tool would be nice to have, i.e., a tool that would incorporate the relevant abstractions / query info from the daemon.
I’d like to feature request a tool to provide this info, e.g., via `ros2 node info --verbose` command.
Maybe a similar query for info belonging to a GID for a topic would be useful too, in case node names collide).

I already skimmed through this repo but I didn’t find any obvious angle of approach to add this myself.
Any suggestions welcome.

Previous discussion:

* https://discourse.ros.org/t/feature-request-provide-cli-to-retrieve-ip-address-a-node-runs-on/28569/

In the linked discussion, the ROS 2 Monitor GUI tool was mentioned that seems to provide a graphical interface for inspection of the running nodes (not sure how vendor-specific this is though):

* https://fast-dds-monitor.readthedocs.io/en/latest/rst/ros/galactic/galactic.html#monitoring-network

fujitatomoya, 2022-12-05:
@moooeeeep thanks for creating issue.

> I recently was in the situation, that two nodes were publishing on the same topic (classroom environment with several turtlebots, two of which with colliding ROS_DOMAIN_ID), and it would have been [...]

i would like to know how this is gonna solve your situation. you want to identify the node based on IP address? probably i am not really sure about the use case and what you really want to use this IP address information in your application.

moooeeeep, 2022-12-06:
> i would like to know how this is gonna solve your situation. you want to identify the node based on IP address? probably i am not really sure about the use case and what you really want to use this IP address information in your application.

Knowing the IP address, I could directly identify and log-in to the offending host and adjust the `ROS_DOMAIN_ID` settings there. Without it, I would need to try all the hosts, to find out which one has it wrong (or am I missing something here?). 

Not sure if it has to be the `node info` verb to query this, or if it would be easier to add a new verb for listing and inspecting DDS participants directly (e.g., `ros2 pub list`, `ros2 pub info ...`).