ros2/ros2cli #790: Subdirectories not installed with ros2pkg's ament_python template

ijnek, 2023-01-14:
Currently, the ament_python template has ``packages=[package_name]``, as seen below:

https://github.com/ros2/ros2cli/blob/cf020c0f0f90ded9aca6ebac806c78a387347a25/ros2pkg/ros2pkg/resource/ament_python/setup.py.em#L8

However, with this configuration, it seems that subdirectories are not installed when running ``colcon build``.

Perhaps it is better to have in the template: 

```python
packages=find_packages(exclude=['test']),
```

like many ros2 python packages do, including most packages in ros2cli do. Happy to raise a PR if others agree with this change.

Related issue: https://github.com/ros-sports/soccer_ipm/pull/7

clalancette, 2023-02-02:
After discussing this, it seems that using `find_packages` is more standard, so we'd be happy to review a PR that changes this.  Note that if we make this change, we'll also have to update the tutorials at https://github.com/ros2/ros2_documentation .

ijnek, 2023-02-03:
Thanks for having a discussion about this, I've gone ahead and created a PR in #801 and in https://github.com/ros2/ros2_documentation/pull/3302 to update the docs.

PR #801 (2023-02-05): Resolve #790
Resolves: #790
commit ad0c2a88 (2023-02-05): resolve #790 (#801)
commit 86ae3930 (2023-05-16): resolve #790 (#801) (#832)

(cherry picked from commit ad0c2a881c42e12840eaff337a5ca9db5bca3b47)
closed 2023-02-05 by gbiggs