ros2/ros2cli #794: topic pub cannot handle multiple `!` lines up in a row

fujitatomoya, 2023-01-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - source build
- Version or commit hash:
  - rolling:HEAD
- DDS implementation:
  - Any
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 topic echo /chatter
data: Hello ROS
---
data: Hello ROS!
---

root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 topic pub --once /chatter std_msgs/msg/String "data: Hello ROS"
Waiting for at least 1 matching subscription(s)...
Waiting for at least 1 matching subscription(s)...
publisher: beginning loop
publishing #1: std_msgs.msg.String(data='Hello ROS')

root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 topic pub --once /chatter std_msgs/msg/String "data: Hello ROS!"
publisher: beginning loop
publishing #1: std_msgs.msg.String(data='Hello ROS!')

root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 topic pub --once /chatter std_msgs/msg/String "data: Hello ROS!!"
ros2 topic pub --once /chatter std_msgs/msg/String "data: Hello ROSros2 topic pub --once /chatter std_msgs/msg/String "data: Hello ROS!""
usage: ros2 [-h] [--use-python-default-buffering] Call `ros2 <command> -h` for more detailed usage. ...
ros2: error: unrecognized arguments: Hello ROS!
```

#### Expected behavior

successfully publishes `ros2 topic pub --once /chatter std_msgs/msg/String "data: Hello ROS!!"`

#### Actual behavior

fails to issue the command line.

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 topic pub --once /chatter std_msgs/msg/String "data: Hello ROS!!"
ros2 topic pub --once /chatter std_msgs/msg/String "data: Hello ROSros2 topic pub --once /chatter std_msgs/msg/String "data: Hello ROS!""
usage: ros2 [-h] [--use-python-default-buffering] Call `ros2 <command> -h` for more detailed usage. ...
ros2: error: unrecognized arguments: Hello ROS!
```

fujitatomoya, 2023-01-27:
@iuhilnehc-ynos any idea?

iuhilnehc-ynos, 2023-01-28:
@fujitatomoya 

It's not a bug. It's just a feature of bash.

`!!` in the bash can call the last command, and please see [the double quotes](https://www.gnu.org/software/bash/manual/html_node/Double-Quotes.html).
```
Enclosing characters in double quotes (‘"’) preserves the literal value of all characters within the quotes, with the exception of ‘$’, ‘`’, ‘\’, and, when history expansion is enabled, ‘!’.
```

We can use `\` or single quotes to avoid this issue.

```
$ ros2 topic pub --once /chatter std_msgs/msg/String "data: Hello ROS\!\!"
$ ros2 topic pub --once /chatter std_msgs/msg/String 'data: Hello ROS!!'
```

fujitatomoya, 2023-01-28:
@iuhilnehc-ynos thanks for the comment. I actually did not realize that, thanks for the information.

closed 2023-01-28 by fujitatomoya