ros2/ros2cli #806: Ros2 doctor fails to call some class functions

JvE-Lely, 2023-02-27:
## Bug report

**Required Info:**

- Operating System:
  - Debian buster
- Installation type:
  - source
- Version or commit hash:
  - Humble 
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  -  N/A

#### Steps to reproduce issue

```
ros2 doctor --report
```

#### Expected behavior

#### Actual behavior
```
ros2 doctor --report
/opt/ros/humble/lib/python3.7/site-packages/ros2doctor/api/__init__.py: 154: UserWarning: Fail to call PackageReport class functions.
/opt/ros/humble/lib/python3.7/site-packages/ros2doctor/api/__init__.py: 154: UserWarning: Fail to call RosdistroReport class functions.
```
and
```
ros2 doctor
/opt/ros/humble/lib/python3.7/site-packages/ros2doctor/api/__init__.py: 118: UserWarning: Fail to call PackageCheck class functions.
/opt/ros/humble/lib/python3.7/site-packages/ros2doctor/api/__init__.py: 118: UserWarning: Fail to call PlatformCheck class functions.
```

#### Additional information
The following packages are installed:
```
 python3-catkin-pkg-modules_0.5.2-1_all.deb
 python3-lark-parser_0.7.2-1osrf~buster_all.deb
 python3-netifaces_0.10.4-1+b1_armhf.deb
 python3-numpy_1.16.2-1_armhf.deb
 python3-rosdistro_0.9.0-100_all.deb
 python3-rosdistro-modules_0.9.0-1_all.deb
 python3-rospkg_1.4.0-100_all.deb
 python3-rospkg-modules_1.4.0-1_all.deb

mjcarroll, 2023-02-27:
I cannot reproduce on jammy, so it may be something `buster` specific in this case.  I will try to spin a docker container to reproduce.

zhuzy-2018, 2023-07-13:
i've encoutered this issue as well using humble on ubuntu 22.04.

fujitatomoya, 2023-07-14:
I cannot reproduce this issue with ubuntu 22.04 and Humble.

```bash
root@tomoyafujita:~# ros2 doctor

All 5 checks passed

root@tomoyafujita:~# ros2 doctor --report
...<snip>
   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-5.15.0-76-generic-x86_64-with-glibc2.35
release          : 5.15.0-76-generic
processor        : x86_64

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : humble
distribution type      : ros2
distribution status    : active
release platforms      : {'debian': ['bullseye'], 'rhel': ['8'], 'ubuntu': ['jammy']}

   TOPIC LIST
topic               : none
publisher count     : 0
subscriber count    : 0
...<snip>
```

zhuzy-2018, 2023-07-14:
> I cannot reproduce this issue with ubuntu 22.04 and Humble.
> 
> ```shell
> root@tomoyafujita:~# ros2 doctor
> 
> All 5 checks passed
> 
> root@tomoyafujita:~# ros2 doctor --report
> ...<snip>
>    PLATFORM INFORMATION
> system           : Linux
> platform info    : Linux-5.15.0-76-generic-x86_64-with-glibc2.35
> release          : 5.15.0-76-generic
> processor        : x86_64
> 
>    QOS COMPATIBILITY LIST
> compatibility status    : No publisher/subscriber pairs found
> 
>    RMW MIDDLEWARE
> middleware name    : rmw_fastrtps_cpp
> 
>    ROS 2 INFORMATION
> distribution name      : humble
> distribution type      : ros2
> distribution status    : active
> release platforms      : {'debian': ['bullseye'], 'rhel': ['8'], 'ubuntu': ['jammy']}
> 
>    TOPIC LIST
> topic               : none
> publisher count     : 0
> subscriber count    : 0
> ...<snip>
> ```

❯ ros2 doctor --report
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/__init__.py: 154: UserWarning: Fail to call PackageReport class functions.
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/__init__.py: 154: UserWarning: Fail to call RosdistroReport class functions.

   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<UP,RUNNING,LOOPBACK>
mtu          : 65536
inet         : 192.168.1.107
inet4        : ['192.168.1.107']
ether        : f8:e4:e3:3f:16:e6
inet6        : ['fe80::9fb8:a835:73c6:38af%wlp82s0']
netmask      : 255.255.255.0
device       : wlp82s0
flags        : 4163<UP,MULTICAST,RUNNING,BROADCAST>
mtu          : 1500
broadcast    : 192.168.1.255
ether        : f8:75:a4:e7:7d:84
device       : enp0s31f6
flags        : 4099<UP,MULTICAST,BROADCAST>
mtu          : 1500

   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-5.15.0-76-generic-x86_64-with-glibc2.35
release          : 5.15.0-76-generic
processor        : x86_64

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   TOPIC LIST
topic               : none
publisher count     : 0
subscriber count    : 0

ZhaoCake, 2023-07-16:
For the outputs of "ros2 doctor", it just because your proxy network may not good. I solve it by use a more quick proxy.

ZhaoCake, 2023-07-16:
And after my test, the outputs of "ros2 doctor --report" is the same.

Cheny5863, 2023-07-16:
i've encoutered this issue as well using humble on ubuntu 22.04.
```
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/__init__.py: 118: UserWarning: Fail to call PlatformCheck class functions.
```

ZhaoCake, 2023-07-16:
> 

My environment is just the same as yours. Do you have tried to use a quick proxy connected to US? The proxy connected to HK doesn't work when I have tested. So maybe a quick proxy to western country will work well. And when I say quick, it means a delay less than 500ms. I even got this warning when I with my proxy on,  just because of the poor quality of networr connection.

> i've encoutered this issue as well using humble on ubuntu 22.04.
> 
> ```
> /opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/__init__.py: 118: UserWarning: Fail to call PlatformCheck class functions.
> ```

My environment is just the same as yours. Do you have tried to use a quick proxy connected to US? The proxy connected to HK doesn't work when I have tested. So maybe a quick proxy to western country will work well. And when I say quick, it means a delay less than 500ms. I even got this warning when I with my proxy on,  just because of the poor quality of networr connection.

MatthijsBurgh, 2023-10-09:
The package check fails, when it can't find a version of a local of req package (see compare_versions). The empty string will result in an InvalidVersion exception. Which should be catched, because the missing pkgs are collected to be logged later on.

MatthijsBurgh, 2023-10-10:
I am running packaging 23.1, which does raise this issue:
```py
>>> import packaging
>>> packaging.__version__
'23.1'
>>> from packaging import version
>>> version.parse('')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.10/dist-packages/packaging/version.py", line 52, in parse
    return Version(version)
  File "/usr/local/lib/python3.10/dist-packages/packaging/version.py", line 198, in __init__
    raise InvalidVersion(f"Invalid version: '{version}'")
packaging.version.InvalidVersion: Invalid version: ''
>>> 
```

fujitatomoya, 2023-10-10:
https://github.com/pypa/packaging/issues/530 removes the deprecated `LegacyVersion` after `v22.0`.

closed 2023-10-17 by clalancette