ros2/ros2cli #809: test_echo_pub.TestROS2TopicEchoPub.test_pub_maxwait_yields failed often recently

iuhilnehc-ynos, 2023-03-02:
## Bug report

log on CI for Linux
```
Error Message
test_echo_pub.TestROS2TopicEchoPub.test_pub_maxwait_yields failed
Stacktrace
======================================================================================================================================================
FAIL: test_echo_pub.TestROS2TopicEchoPub.test_pub_maxwait_yields
------------------------------------------------------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jenkins-agent/workspace/ci_linux/ws/install/launch_testing/lib/python3.10/site-packages/launch_testing/markers.py", line 57, in _wrapper
    return func(self, *args, **kwargs)
  File "/home/jenkins-agent/workspace/ci_linux/ws/src/ros2/ros2cli/ros2topic/test/test_echo_pub.py", line 258, in test_pub_maxwait_yields
    assert future.done()
AssertionError: assert False
 +  where False = <bound method Future.done of <rclpy.task.Future object at 0x7f1dd13c2500>>()
 +    where <bound method Future.done of <rclpy.task.Future object at 0x7f1dd13c2500>> = <rclpy.task.Future object at 0x7f1dd13c2500>.done
```

https://ci.ros2.org/job/ci_linux/18139/testReport/junit/ros2topic.ros2topic.test/test_echo_pub/test_echo_pub/
https://ci.ros2.org/job/ci_linux/18137/testReport/junit/ros2topic.ros2topic.test/test_echo_pub/test_echo_pub/
https://ci.ros2.org/job/ci_linux/18133/testReport/junit/ros2topic.ros2topic.test/test_echo_pub/test_echo_pub/
https://ci.ros2.org/job/ci_linux/18129/testReport/ros2topic.ros2topic.test/test_echo_pub/test_echo_pub/
https://ci.ros2.org/job/ci_linux/18128/testReport/junit/ros2topic.ros2topic.test/test_echo_pub/test_echo_pub/
https://ci.ros2.org/job/ci_linux/18126/testReport/junit/ros2topic.ros2topic.test/test_echo_pub/test_echo_pub/

fujitatomoya, 2023-03-02:
@iuhilnehc-ynos this is happening because of https://github.com/ros2/ros2cli/pull/800?

iuhilnehc-ynos, 2023-03-02:
I think so, but I haven't dived into the deep.

iuhilnehc-ynos, 2023-03-02:
The message can't be reliable to receive without a correct QoS setting.

Crola1702, 2023-03-02:
FYI: @arjo129