ros2/ros2cli #820: return raw parameter content without type information

christian-rauch, 2023-04-12:
## Feature request

#### Feature description

`ros2 param get` needs an option to return the raw content of a parameter. Currently, it prepends the parameter content with the value type.

For example:
```sh
ros2 param get /move_group robot_description
```
returns:
```
String value is: <?xml version="1.0" ?> [...] <robot name="panda"> [...] </robot>
```

It would be much more useful if `get` would return the raw content of the parameter, i.e. the URDF string in the case above without `String value is: ` prepended, such that it can be parsed by another tool.

#### Implementation considerations
This can be implemented by a new flag for `get`, that either adds or omits the type information (`String value is: `). IMHO, it makes more sense to omit the type by default and only add type information when the "verbose type" flag is set.

fujitatomoya, 2023-04-12:
@christian-rauch thanks for creating issue.

are you specifically requesting the format with xml? i do not think xml is not supported even with `load` or `dump`.

> It would be much more useful if get would return the raw content of the parameter, i.e. the URDF string in the case above without String value is:  prepended, such that it can be parsed by another tool.

can you share the actual use case if you have with examples?

btw, you can `dump` the parameters in yaml file.

```yaml
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param dump /parameter_blackboard
/parameter_blackboard:
  ros__parameters:
    qos_overrides:
      /parameter_events:
        publisher:
          depth: 1000
          durability: volatile
          history: keep_last
          reliability: reliable
    use_sim_time: false
```

iuhilnehc-ynos, 2023-04-13:
@christian-rauch 

I think you could try to use the parameter  `--hide-type`.

```
$ ros2 param get -h
usage: ros2 param get [-h] [--spin-time SPIN_TIME] [-s] [--no-daemon] [--include-hidden-nodes] [--hide-type] [--timeout N] node_name parameter_name

Get parameter

positional arguments:
  node_name             Name of the ROS node
  parameter_name        Name of the parameter

options:
  -h, --help            show this help message and exit
  --spin-time SPIN_TIME
                        Spin time in seconds to wait for discovery (only applies when not using an already running daemon)
  -s, --use-sim-time    Enable ROS simulation time
  --no-daemon           Do not spawn nor use an already running daemon
  --include-hidden-nodes
                        Consider hidden nodes as well
  --hide-type           Hide the type information
  --timeout N           Wait for N seconds until node becomes available (default 1 sec)
```

https://github.com/ros2/ros2cli/blob/f31cd91bb52c90b67e9b9e76aaa56e6eced0803b/ros2param/ros2param/verb/get.py#L102-L105

christian-rauch, 2023-04-13:
> I think you could try to use the parameter `--hide-type`.

I overlooked that flag. Thanks.

This is exactly what I was looking for, i.e.:
```sh
ros2 param get --hide-type /move_group robot_description
```
gives the raw content (URDF as XML string) without the type information prepended. IMHO, this should be the default.

closed 2023-04-13 by christian-rauch