ros2/ros2cli #842: Load a message or request from a yaml file (or standard input)

ymd-stella, 2023-07-16:
## Feature request

#### Feature description

I want to load a message/request from a yaml file, not from the command line. The benefits of this would be similar to #753.

Targets are

- ros2 topic pub
- ros2 service call
- ros2 action send_goal

In my opinion, it would be neat to be able to do the following

`cat foo.yaml | ros2 topic pub <topic_name> <msg_type>`

#### Implementation considerations

This maybe conflicts with the functionality of #753. Equivalent functionality may be achieved with `cat foo.yaml | vipe | tee foo.yaml | <topic_name> <msg_type>`.

fujitatomoya, 2023-07-24:
this sounds reasonable to me. it would be useful.

ymd-stella, 2023-07-24:
I noticed that the last argument is optional. For example, the following is valid.

```
ros2 topic pub foo std_msgs/Empty
```

Should we add an option like `--stdin` to keep the default behavior?

```
cat foo.yaml | ros2 topic pub --stdin <topic_name> <msg_type>
```

clalancette, 2023-07-24:
> Should we add an option like `--stdin` to keep the default behavior?

Yes, definitely.

ymd-stella, 2023-07-25:
> This maybe conflicts with the functionality of https://github.com/ros2/ros2cli/pull/753. Similar functionality may be achieved with cat foo.yaml | vipe | tee foo.yaml | ros2 topic pub <topic_name> <msg_type>.

Using vscode, this could be achieved with the following command.

```bash
export EDITOR="code --wait"
cat foo.yaml | vipe | tee foo2.yaml | ros2 topic pub foo std_msgs/Header --stdin; mv tmp.yaml foo.yaml
```

jrutgeer, 2023-07-28:
Interesting.

I have never used ROS 1, but [according to the wiki](http://wiki.ros.org/rostopic#rostopic_pub) it seems that `rostopic pub` supported piped input, but also had a `-f` flag to read from file:

```
$ rostopic echo chatter > chatter.bagy
Collect messages, then Ctrl-C
$ rostopic pub -f chatter.bagy bar std_msgs/String
```

Given the problem statement:

> I want to load a message/request from a yaml file, not from the command line.

Would you consider to also add the `-f` flag to `ros2 topic pub`?

PR #844 (2023-08-03): Load a message/request/goal from standard input
Resolve #842

Add option `--stdin` to pass yaml from standard input.
The last positional argument and the new option `--stdin` are mutually_exclusive_group.

The change requires that the send_goal argument `goal` be optional.
closed 2023-08-03 by clalancette