ros2/ros2cli #843: ros2 topic hz perform worse than c++ code, also worse than result in another docker container

Chenxintang, 2023-07-18:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu18.04
- Installation type:
  - source
- Version or commit hash:
  - humble
- DDS implementation:
  - FastDDS

#### Steps to reproduce issue

- run two docker container using the following docker-compose.yml
```yaml
container1:
    image: image_name
    network_mode: "host"
    cpuset: 2-9
    cpu_shares: 2048
    shm_size: 4G
    container_name: container1
    runtime: nvidia
    privileged: true
    environment:
      - NVIDIA_VISIBLE_DEVICES=all
    logging:
      options:
        max-size: "${DOCKER_LOGS_MAX_SIZE:-100M}"
```
```yaml
container2:
    image: image_name
    network_mode: "host"
    privileged: true
    cpuset: 12-17
    cpu_shares: 2048
    shm_size: 2G
    container_name: container2
    environment:
      - NVIDIA_VISIBLE_DEVICES=all
    runtime: nvidia
    logging:
      options:
        max-size: "${DOCKER_LOGS_MAX_SIZE:-100M}"
```
- publish `sensor_msgs::msg::PointCloud2` type message with frequency 10hz in container1
- run subscriber and process data c++ code in container2, record the frequency subscription callback called
- use `ros2 topic hz` tool follow the frequency of topic in both containers
#### Expected behavior
- In both containers, `ros2 topic hz` shows the same frequency which is around 10hz
- Log of c++ code in container2 shows that pointcloud2 messages arrive each 100ms 
#### Actual behavior
- In container1 terminal, `ros2 topic hz` shows frequency slightly under 10hz
[image]
- In container2 terminal, `ros2 topic hz` shows frequency highly under 10hz, verity from 7hz to 9hz
[image]
- In log of container2, c++ code receive message each 100ms stably
[image]

#### Additional information
Since code in both container are quite complicate, I can't paste the code here.
As discussed in the blow question, ros2 topic hz result is lower than c++ code because the qos used is different. But why the hz results in different containers turn out to be different?
>https://answers.ros.org/question/350753/ros2-topic-hz-provides-wrong-rate-for-larger-msgs/
Also, since ros2 topic hz result is not reliable, is therea a better way to track the transportaion stability between many nodes?

fujitatomoya, 2023-07-19:
> Ubuntu 18.04

Humble supports Ubuntu 22.04 as tier 1 platform.
I am not sure if Humble can built with Ubuntu 18.04. (btw, Ubuntu 20.04 is community support.) 
see more details for https://docs.ros.org/en/humble/Releases/Release-Humble-Hawksbill.html#supported-platforms.

> In container1 terminal, ros2 topic hz shows frequency slightly under 10hz

i believe that in this case, shared memory transport (not zero copy via LoanedMessage) can be enabled as localhost communication. this could be one of the benefits for the performance.

> In container2 terminal, ros2 topic hz shows frequency highly under 10hz, verity from 7hz to 9hz

inter-process communication via network interface (in the case of docker, via bridge) would be one of the reason for this.

> In log of container2, c++ code receive message each 100ms stably

there could be several differences, like language advantage for performance, `rclpy pybind` performance overhead.

above is my assumption, i did not test and analyze the performance difference. hopefully this helps.

fujitatomoya, 2025-03-12:
There are a few issues here, I suggest the following.

- Performance issue of rclpy. This is well known issue that rclpy is not well-performative compared to rclcpp. (this is out of scope this issue.)
- [No need to deserialize the message via hz and bw unless necessary](https://github.com/ros2/ros2cli/issues/983), minor enhancement to make it more efficient for hz and bw sub-commands.
- add more description about how ros2 topic hz/bw (or anything else like these based on subscription) works. It might not show the expected frequency set on publisher, it just show `what subscription/receiver actually get`. and this is affected by QoS configuration and platform computation resource. (ros2 documentation can be also updated with quick note like this.)

TanayK07, 2026-10-18:
adding numbers for the docs point in fujitatomoya's list, since #1005 fixed the deserialization but the subscription cost is still there. measured on ros:humble (ros2cli 0.18.13+, so hz already takes raw messages), cyclone, one 16 core box, 10 paired trials:

- hz on one 100 KB topic at 50 Hz used 7.1% of a core by itself. echo on the same topic into /dev/null used 31%.
- hz on a topic that only had intra-process subscribers made the publishing process 52% more expensive, because it now had to serialize every message for hz. the rate hz printed was right, the process it described just wasn't the one running before.

so for the docs, something like "hz is a regular subscriber: it shows what a new subscriber receives, it costs cpu per watched topic, and on an intra-process topic it makes the publisher serialize" would cover it. happy to open a docs pr if that's useful.

bench script and raw csv: https://github.com/TanayK07/ros2_pulse/blob/main/bench/RESULTS.md#observer-effect-what-watching-a-topic-with-the-stock-cli-costs-2026-08-23 (that's my tool's repo, it counts inside the process instead of subscribing)