ros2/ros2cli #855: Backport timeout options on humble

tokoro10g, 2023-09-25:
## Feature request

#### Feature description
- Backport options such as `ros2 topic echo --time-out` on humble

#### Implementation considerations
Is there any policy for backporting? My opinion is that these basic tools should have the same interface across the distributions until it reaches the EOL. What do you say?

fujitatomoya, 2023-09-26:
you can see https://docs.ros.org/en/rolling/How-To-Guides/Package-maintainer-guide.html#backporting-to-released-distributions as guideline.

AFAIK, it also depends on the behavior change that could affect the application, tools and so on.

adding command line options can be considered as behavior change. but on this one, i do not see any of that, so it would be probably okay to go? that is not i am 100% sure, so lets keep this open for more feedback.

Related PRs,

- https://github.com/ros2/ros2cli/pull/792
- (https://github.com/ros2/ros2cli/pull/695)

CC: @ros2/team @clalancette @wjwwood @audrow

tokoro10g, 2023-09-26:
I would also like these backported:
- https://github.com/ros2/ros2cli/pull/800
- https://github.com/ros2/ros2cli/pull/802

clalancette, 2023-09-26:
> Is there any policy for backporting? My opinion is that these basic tools should have the same interface across the distributions until it reaches the EOL. What do you say?

The general policy is to only backport as things are requested, and as feasible.  Because of our strong API and ABI guarantees in the core, backports are often non-trivial (though they tend to be easier with Python).

For the ones you've requested:

* #792 - does not change API or ABI, so should be safe
* #695 - does not change API or ABI, so should be safe
* #800 - does not change API or ABI, so should be safe
* #802 - does seem to change API, so probably needs to be refactored a bit for a backport to Humble

@tokoro10g if you'd like to backport these (keeping the current API), we'd be happy to review them